OctoPrint/OctoPrint #3596: [Request] Ability to move files from the root file list into a folder

NameOfTheDragon, 2020-06-09:
**Feature Description: File Management**

I like to keep printed files on the server but the file list gets cluttered rather quickly. I create folders to group together files related to a project. When I print, I typically print directly from my slicer, which always puts the uploaded file in the root file list. I would like to be able to come back later and move these files into my project folders.

**Suggested Solution**

Two alternatives suggest themselves.

1. A button next to each file "Move to folder...". When clicked, a folder browser os displayed which allows me to select the destination folder.

2. A checkbox next to each file which allows multiple files to be selected, then a "Bulk Actions" drop-down at the top or bottom of the file list which would include "Move to Folder...". (Other useful items such as delete could be included here).

jneilliii, 2020-06-09:
There's a file [manager plugin](https://plugins.octoprint.org/plugins/filemanager/) as well as a [DeleteMoveAfterPrint](https://plugins.octoprint.org/plugins/DeleteAfterPrint/) plugin.

tcurdt, 2020-06-19:
Unfortunately the manager plugin has "WARNING Plugin actually not in active development!" printed in big letters on it's github page.

It's also very weird to have "Create Folder" supported in core, but you can't really do anything with the folder.

The DeleteMoveAfterPrint is a hole different usecase.

foosel, 2020-06-19:
I didn't really see a need building a file manager into OctoPrint when @Salandora already did a pretty good job with providing one with his plugin (which he developed alongside the addition of the folder support).

However, @Salandora, would you mind if I "take the filemanager off your hands" and simply bundle it into OctoPrint? I could continue to maintain it that way, it would be off your plate, and tickets like this one here would also stop 😅

fieldOfView, 2020-06-19:
Though I think the file manager is great as a plugin I'm not a fan of the plugin myself. It duplicates a lot of functionality that is already available in the file lister. I would much rather see the file lister enhanced with some simple drag-and-drop functionality for moving files into subfolders. And that's the point: people have preferences and having a choice of plugins for that is great.

Keep OctoPrint simple and don't bundle plugins that limit choices.

tcurdt, 2020-06-19:
If the core file lister allows for folder creation I would think it should also allow for simple moving of files. That's what I would prefer, too. But unless someone is willing to spend the time of merging the two, bundling might be the most pragmatic way forward. And if there is no other file manager the idea of choice is an illusion anyway.

tcurdt, 2020-06-19:
If it wasn't for the `.metadata.json` I would just ssh into my pi and move the files via shell.

jneilliii, 2020-06-19:
I don't think it would be too difficult to implement the necessary changes to achieve this in the standard sidebar file list, but would need direction on how you'd expect it to work from a UI perspective. Click a button and get a popup with folder list to select destination seems to be the original intent. I would approach it as a plugin though rather than changing core functionality.

tcurdt, 2020-06-19:
@jneilliii I personally was actively searching for a "move" icon that then let me pick a folder where to move to. And was stunned not to find such icon. I then searched, found the File Manager and ended up here. So - I was exactly looking for the UI you described.

foosel, 2020-06-19:
I'd be happy to merge a PR that adds a move button

jneilliii, 2020-06-19:
Which branch should that be based on for a PR?

foosel, 2020-06-19:
It's an improvement, so `maintenance`. API methods are all there, it's just UI that's missing.

jneilliii, 2020-06-20:
Hopefully it's close enough that there's not too many changes necessary to meet your standards @foosel.

jneilliii, 2020-06-23:
Yay, looks like there weren't too many changes necessary.

foosel, 2020-06-23:
Just some changes to the UX really, and elimination of one potentially expensive request.

Salandora, 2020-07-20:
@foosel sorry for the late reply, sure you can bundle the fileManager if you like.
I'll still try to improve it and get rid of some code mess I made back in the days.

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel