OctoPrint/OctoPrint #3597: [Request] Ultimaker 3 Plugin

tgrobeln, 2020-06-09:
Dear OctoPrint Community,

I'm currently an automation electrician apprentice in a German company.  

During Corona I've been working with 3D printing. We have made ear mask savers and doorhandel attachments - so we could use our elbows instead of our hands to open the doors. We have used PLA for the ear mask savers and ABS for the doorhandel attachments.

During that time, we had also some trouble printing - that's why I've searched the internet and found Octoprint. I knew what a raspberry Pi was and always wanted to work with it, but I couldn't find a great project until now.

I wanted to connect an Ultimaker 3 with Octoprint and there is also where the problems starts. 
The Ultimaker 3 doesn't support a serial connection. I know one can connect an Ultimaker 3 with Cura via the ethernet port on the back of the 3D printer. But only for the sake trasnmitting the data from Cura to the 3D printer via an ethernet connection and receiving some footage from the integrated webcam -  that has an resolution of a pot of mashed potatoes - is not worth the effort.

Is there any possibility to creat a plugin so that the Octiopi can be connected with the Ultimaker 3 via an ethernet connection?

Did anybody ever consider to programm such a plugin? I'm really a rookie in programming and I don't have any kind of clue how complex it will be and if it's even possible to start such an project.

We would love to make timelaps videos of our prints and show a monitoring of printing with g-code on technology exibitions to attackt students to start an apprenticeship.

Thank you for reading my request and your answers

fieldOfView, 2020-06-09:
The UM 3 (and the newer S5 and S3) internally combine a linux-based small board computer with a Marlin-based controller board. That SBC (an Olimex A20 derivative) not only handles queueing and sending gcode to the Marlin controller board, but it also handles some of the higher logic of switching "printcores". The software stack that does this (named "Ultimaker Connect") is not open source, so it is hard to develop against and even harder to only bypass the parts that would need bypassing to somehow have OctoPrint control it.

fieldOfView, 2020-06-09:
I forgot to answer your question:

> Is there any possibility to creat a plugin so that the Octiopi can be connected with the Ultimaker 3 via an ethernet connection?

Realistically? No.

What is it that you think you'll gain by using OctoPrint instead of Ultimaker Connect?