OctoPrint/OctoPrint #3598: Can't install plugins using SSH

VASAPOL, 2020-06-11:
pi@octopi:~ $ cd OctoPrint-Lcd1602 && python setup.py install
Traceback (most recent call last):
  File "setup.py", line 5, in <module>
    from setuptools import setup
ImportError: No module named setuptools
pi@octopi:~/OctoPrint-Lcd1602 $

VASAPOL, 2020-06-11:
I already contact plugin developer he said I need to install setup tools and don't know how to resolve this problem.

Plug in: Lcd-1602

fieldOfView, 2020-06-11:
This is not a bug in OctoPrint. To get support for your OctoPrint installation and OctoPi, please use the community forum instead:
https://community.octoprint.org

If you do, please include more information about your environment. I *think* you may be using OctoPrint via the OctoPi image. In that case you have to activate the virtualenv before you can install plugins like that.

closed 2020-06-11 by fieldOfView