OctoPrint/OctoPrint #3599: Restore from backup request fails

Uko, 2020-06-11:
#### What were you doing?

1. Go to Settings -> Backup and Restore
2. Choose a backup file in the "Restore from uploaded backup" section
3. Click "Upload and Restore" and click "Proceed"
4. I see a "Uploading backup, this can take a while. Please wait..." modal window and nothing ever changes.

#### What did you expect to happen?
I expected the restore functionality to work (idk, load the file and restore the setup from it) .

#### What happened instead?
In a web(network) inspector, I see that requests to the `/plugin/backup/restore` path fail. For two of them, I got a status code `502 Bag Gateway`, for one I didn't see any status code (I checked in the Chrome dev tools).

The request fails imediately after it's sent.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

Octopi 0.17.0

#### Printer model & used firmware incl. version

It's a fresh install, I don't have any printer setup yet.

#### Browser and version of browser, operating system running browser

Safari 13.1
Google Chrome  83.0.4103.97

macOS 10.15.4

#### Link to octoprint.log

https://pastebin.com/ZberWHNx

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

https://pastebin.com/FVdhnvXJ

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tedder, 2020-06-11:
There are no indications in that log of an error or any action, really. Are you sure the log covers the time period when you are doing a 'restore'? Please do it again while tailing the log to see what comes out at the moment of restore.

Uko, 2020-06-11:
Yes, I was tailing the log while trying to do a restore and nothing gets appended after the request is sent. Is there any way I can debug this better?

foosel, 2020-06-11:
If it's a Bad Gateway, it happens before OctoPrint itself, between the reverse proxy and OctoPrint to be precise. So either it's an issue with haproxy/the image, or something else before even that. Doesn't look like it's something that could be solved in OctoPrint, and last I checked backup restore worked just fine in general.

Any additional network hops between your browser and the pi?

foosel, 2020-06-18:
@Uko waiting for information from you here.

closed 2020-06-23 by foosel