OctoPrint/OctoPrint #3604: [Request] Include All Parameters of REST API Upload in Upload Event Payload

jneilliii, 2020-06-13:
**Is your feature request related to a problem? Please describe.**
For most of my power plugins many people have requested auto-powering on the printer when a file is uploaded. I think this is mainly associated with PrusaSlicer as the Cura Connection plugin already handles managing it using internal plugin APIs. It would be nice to be notified of these provided options in the Event payload so plugin's could react accordingly. As it stands now when using either `FileAdded` or `Upload` events to act on a new file there is no way to know if it was set to automatically be selected or printed. 

**Describe the solution you'd like**
Receive all parameters from the file upload REST API in the event payload to allow plugins to use that information for decision making.

**Describe alternatives you've considered**
I thought of utilizing the `FileSelected` or `FileAdded` event, but based on the docs it doesn't appear to include a flag for automatically starting the print either. In my use case if the printer's controller is not back-powered by USB there is no way to achieve powering on the printer, waiting a bit, connecting, and starting a print when a file is uploaded without doing it for every file uploaded, including those from the UI.

foosel, 2020-06-18:
Implemented by the above commit, ready for 1.4.1

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel