OctoPrint/OctoPrint #3606: [Request] Improve plugin manager to include more info like installs

danielo515, 2020-06-17:
Hello, and thanks for this fantastic tool octopint is, I'm very grateful it exists.
Today I'm here to make a feature request/suggestion.

**Is your feature request related to a problem? Please describe.**
I think the plugin manager can be improved by adding some extra bits of information that I will describe below

**Describe the solution you'd like**
The plugin manager is probably the main interface people use to install and discover new plugins. However, it lacks certain key information pieces that are very important to decide if you want to install a plugin or not. A list that comes to my mind includes:

- When the plugin was last updated (date)
- How many times a plugin has been installed (popularity)
- When the plugin was first registered (maturity)
- If the plugin is deprecated or no longer maintained

Those four points are key to decide if you want to install a plugin, or they will help you decide between two plugins (the most popular, the most up to date, the most mature...)

I don't think anything else is required, and since this information is already available on octoprint page, It should be possible

**Describe alternatives you've considered**
The alternative is to keep it as it is

**Additional context**
As I said, this will really improve the user experience on the plugin manager. Aditionally sorting by each field can be implemented, but that is a different story.

foosel, 2020-06-17:
Most of that is actually already prepared and ready for 1.4.1 ;)

foosel, 2020-06-19:
[image]

[image]

Ready for 1.4.1

danielo515, 2020-06-19:
That is fantastic!!! Is there an alternative to use text instead of icons? Not everybody (including me) will grasp the meaning of those. A tooltip will also be enough.

foosel, 2020-06-19:
All of those have tooltips. Text would make it become too much.

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel