OctoPrint/OctoPrint #3610: Logitech C270 webcam stream not loading inconsistency

JamieS01, 2020-06-20:
#### What were you doing?

1. Visit octopi.local (or via ip address)
2. Navigate to control tab
3. Look at the webcam stream

#### What did you expect to happen?

For the webcam stream to load and update for more than a few seconds.

#### What happened instead?

Webcam stream starts working for a second or two after reboot, then the stream window goes black. Shortly after, it displays "webcam stream loading" and then eventually "webcam stream not loaded"

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.4.0

#### Operating System running OctoPrint

OctoPi 0.17.0 on Rasperry Pi 4 4gb

#### Printer model & used firmware incl. version

Creality Ender 3 with customised Marlin 2.0 firmware and SKR Mini E3 mainboard (shouldn't affect anything but no harm in including)

#### Browser and version of browser, operating system running browser

Google Chrome Version 83.0.4103.106 (Official Build) (64-bit), Windows 10 Pro

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/4808540/octoprint.log)

#### Link to contents of terminal tab or serial.log

[serial.log](https://github.com/OctoPrint/OctoPrint/files/4808541/serial.log)
(Ignore at the end - I turned my printer off)

#### Link to contents of Javascript console in the browser

(Not sure how)

#### Screenshot(s)/video(s) showing the problem:

(Should be quite self explanatory but I am happy to create a video)

foosel, 2020-06-20:
Sorry, this is the wrong place for this. OctoPrint doesn't actually manage your webcam, it merely embeds stream URLs in its web interface.

I suggest to open a support topic on the [community forums](https://community.octoprint.org).

closed 2020-06-20 by foosel

JamieS01, 2020-06-20:
> Sorry, this is the wrong place for this. OctoPrint doesn't actually manage your webcam, it merely embeds stream URLs in its web interface.
> 
> I suggest to open a support topic on the [community forums](https://community.octoprint.org).

My apologies! I will do just that, thanks