OctoPrint/OctoPrint #3611: Lcd displaying print finished

Commiccannon, 2020-06-21:
#### What were you doing?

1. Flashed Marlin 2 after having Klipper running for some time and now LCD after about 2 mins will go to the print finished screen

#### The LCD should have continued on the print screen

#### The LCD shows Print finished

#### 1.4.0

#### Octopi 0.17

#### Ender3 with Marlin 2.0

#### Chrome running on windows

#### Link to octoprint.log
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/4808994/octoprint.log)

#### Link to contents of terminal tab or serial.log
[serial.log](https://github.com/OctoPrint/OctoPrint/files/4808993/serial.log)

#### Screenshot(s)/video(s) showing the problem:
[image]
Sorry for the line on the left that dosnt work

I have read the FAQ.

jneilliii, 2020-06-21:
That's not an octoprint issue.

fieldOfView, 2020-06-21:
This issue tracker is strictly for bugs in OctoPrint. This seems to be a firmware issue. Perhaps you can get better support at the OctoPrint community forum (https://community.octoprint.org), which is more suitable for support requests than this issue tracker. But you may also want to check out fora that are more catered towards Creality printers and/or Marlin. 

This issue can be reopened if you convince us OctoPrint is doing something wrong.

closed 2020-06-21 by fieldOfView