OctoPrint/OctoPrint #3618: [Request] Ability to support multiple printers

posicat, 2020-06-29:
**Is your feature request related to a problem? Please describe.**
I've setup multiple printers manually, but it's a pain to do every time a new installed needs to be loaded on the SD.

**Describe the solution you'd like**
It would be great if the instances of the video and octoprint were setup in such a way that there was a simple command line option to add a second instance of Octoprint.  Especially on the new Pi 4's there's more than enough resources to run 3 or 4 at once.  Myself I have 2 printers, soon to have a 3rd.

**Describe alternatives you've considered**
I don this manually, but it's not a fun process.  And it seems a waste to buy a second Pi just to run a second instance.

foosel, 2020-06-29:
This is not going to happen any time soon as the challenges with the current architecture and the expected support overhead are enormous.

Moorviper, 2020-06-29:
You can use https://github.com/paukstelis/octoprint_deploy to setup multiple instances.

jneilliii, 2020-06-29:
Interesting script @Moorviper, does it clear API keys, salt ,etc.?

Moorviper, 2020-06-29:
No this is something you have to do manually.

tedder, 2020-06-29:
[Here's the community-supported megathread](https://community.octoprint.org/t/multiple-printers-per-server-megathread/7778). It isn't supported (by Gina et al) for a good reason. If you are _really_ motivated, you could consider making a variant of OctoPi that sets things up.

closed 2020-06-29 by tedder

fieldOfView, 2020-06-29:
> If you are really motivated, you could consider making a variant of OctoPi that sets things up.

You could call it HexaDecaPi