OctoPrint/OctoPrint #3619: Non-zero exit code is not returned on CLI failure

cp2004, 2020-06-29:
#### What were you doing?

Using https://github.com/cp2004/Octoprint-Upgrade-To-Py3
Run backup plugin's CLI command `octoprint plugins backup:backup --exclude timelapse --exclude uploads`

#### What did you expect to happen?
On a failure of the command, it would return non-zero exit code

#### What happened instead?
My `subprocess.run` command with `check=True` found no issue, as exit code was 0

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPrint
1.4.0

#### Operating System running OctoPrint
OctoPi/Linux/Windows 10. Not platform specific

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
N/A

#### Link to octoprint.log
_No log generated when using CLI commands._

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

Link to discussion of the issue: https://community.octoprint.org/t/upgrade-your-octoprint-install-to-python-3/21232/3

foosel, 2020-07-17:
Hm... I'm confused. 

I just manually added an exception into the backup creation handler, and that makes the return code decidedly non-zero:

```
(devenv)
Gina@VIN MINGW64 /d/Code/OctoPrint/OctoPrint (maintenance)
$ octoprint plugins backup:backup --exclude timelapse --exclude uploads
Initializing settings & plugin subsystem...
Creating backup at octoprint-backup-20200717-163038.zip, please wait...
Traceback (most recent call last):
  File "C:\Devel\OctoPrint\devenv\Scripts\octoprint-script.py", line 11, in <module>
    load_entry_point('OctoPrint', 'console_scripts', 'octoprint')()
  File "d:\code\octoprint\octoprint\src\octoprint\__init__.py", line 629, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "c:\devel\octoprint\devenv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\devel\octoprint\devenv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\devel\octoprint\devenv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\devel\octoprint\devenv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\devel\octoprint\devenv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\devel\octoprint\devenv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "d:\code\octoprint\octoprint\src\octoprint\plugins\backup\__init__.py", line 392, in backup_command
    datafolder=datafolder)
  File "d:\code\octoprint\octoprint\src\octoprint\plugins\backup\__init__.py", line 655, in _create_backup
    raise ValueError("AAAAh")
ValueError: AAAAh
(devenv)
Gina@VIN MINGW64 /d/Code/OctoPrint/OctoPrint (maintenance)
$ echo $?
1
```

Do you have any specific reproduction steps where backup creation fails but a zero code is returned?

cp2004, 2020-07-17:
Hmm. I too am confused. The issue came from the Py 3 Upgrade script when someone on the forum said it failed for them. The strange thing was that although the backup seemed to fail in the script, it worked outside of the script. 

The error they got was no such file, so I thought it must have been that the backup failed some time between 'Creating backup at xxx.zip' which is what the script looks for (and picked up correctly) and OctoPrint actually creating it. At the time it was using `subprocess.run` with `check=True` which raises an exception if there is not a zero exit code. (Now using `subprocess.PIPE` and `.poll()`)

Looking at the post on the forum, it also seems the user made a backup before running the script. I will give this a test, but the can you run more than one backup at a time? (One from web, one cli?) At the time I couldn't think of anything as it worked 2 hours later for them once I got back on the forum. That is the _only_ thing I can think of....

Looking at the source now, I see that it would raise an exception unless you tell it not to (I think). Will have to give it another test. 

Hopefully that makes sense, will give it a play now.

cp2004, 2020-07-17:
Yep, that's it! I am so happy I found this............ It was on my mind for ages once that post came through!

Start a backup from web interface, then try and start one from the command line - 2nd one fails, but with a 0 code. Will have to check it is not the plugin causing that, but on my clean dev install the backup was too quick. Will probably shove a bunch of timelapses in there and see what happens

```
(oprint) pi@Charlies3DPrinter:~ $ python -m octoprint plugins backup:backup --exclude timelapse --exclude uploads
Initializing settings & plugin subsystem...
Initializing GcodePositionProcessor V1.0.1 - Copyright (C) 2019  Brad Hochgesang...Python 3+ Detected...complete
2020-07-17 16:40:32,752 - octolapse.__init__ - INFO - Release mode detected.
Creating backup at octoprint-backup-20200717-164032.zip, please wait...
Done.
Backup located at /home/pi/.octoprint/data/backup/octoprint-backup-20200717-164032.zip
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.7/logging/__init__.py", line 2039, in shutdown
    h.close()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
(oprint) pi@Charlies3DPrinter:~ $ echo $?
0
(oprint) pi@Charlies3DPrinter:~ $
```

cp2004, 2020-07-17:
UPDATE on the above. I got over excited. It actually did create a backup. Now I need to go and check again........

foosel, 2020-07-22:
This only made me more confused XD

cp2004, 2020-07-22:
Haha I haven't been able to reproduce this since. Got exited seeing the error, but the backup is successful so that was a plugin issue. To be fair, I never had this issue myself and it was the only logical way I could see the script fail. There is now an extra check in there to make sure it exists, so hopefully don't get this again. Unless you have anything else to add, I would close this

foosel, 2020-07-22:
In that case I'm closing, as I've also not been able to see anything suspicious in the code. Thanks ^^

closed 2020-07-22 by foosel