OctoPrint/OctoPrint #3620: Please handle filament runout other than "paused->timeout"

AndKe, 2020-06-29:
This is the current experience of filament runout:

```
Send: N26 G1 X248.200 Y56.325 E23.27269*84
Recv: ok
Send: N27 G1 X250.274 Y56.722 E23.46472*87
Recv:  T:210.20 /210.00 B:59.98 /60.00 @:28 B@:0
Recv: echo:busy: processing
Recv: //action:paused filament_runout 0
Recv:  T:210.27 /210.00 B:60.04 /60.00 @:27 B@:0
Recv: echo:busy: processing
Recv: ok
Send: N28 G1 X252.056 Y57.855 E23.65675*82
Recv: //action:paused
Printer signalled that it paused, switching state...
Changing monitoring state from "Printing" to "Pausing"
Recv:  T:210.35 /210.00 B:59.99 /60.00 @:26 B@:127
Recv: echo:busy: processing
Recv:  T:210.31 /210.00 B:60.08 /60.00 @:26 B@:127
Recv: echo:busy: processing
Recv:  T:210.31 /210.00 B:60.30 /60.00 @:26 B@:0
Recv: echo:busy: processing
Recv:  T:210.12 /210.00 B:60.28 /60.00 @:29 B@:0
..............................(and about a minute or so  later.........)
Recv: echo:busy: processing
Recv:  T:210.31 /210.00 B:60.19 /60.00 @:28 B@:0
Recv:  T:210.00 /210.00 B:60.34 /60.00 @:34 B@:0 W:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Pausing" to "Paused"
Send: M105
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Paused" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

```
The job is correctly paused (while the Marlin firmware would wait for filament change)  - but then the communication stops and the print is killed loooong before one can change the filament.

jbbandos, 2020-07-08:
Did you enable HOST_ACTION_COMMANDS and HOST_PROMPT_SUPPORT in Configuration_adv.h in marlin? I thought you needed those to be able to handle filament runout in octoprint.

jbbandos, 2020-07-08:
#### What were you doing?

1.Printing normally (octoprint, Ender 3 pro, marlin 2.0.x fix just cloned from github, configured and compiled)
2. Filament runout sensor tripped
3. Octoprint shows dialog informing of filament runout, times out while printer is reversing the filament, closes communication and aborts print

#### What did you expect to happen?
A pause for filament replacement, a button to press when it has finished, followed by a resume printing
#### What happened instead?
A communication timeout and print aborted
#### Did the same happen when running OctoPrint in safe mode?
Also got a communication timeout and a aborted print, no popup

#### Version of OctoPrint
OctoPrint 1.4.0

#### Operating System running OctoPrint
OctoPi 0.16.0 (raspberry Pi 3)

#### Printer model & used firmware incl. version
Ender 3 Pro, Marlin bugfix-2.0.x (Jul  8 2020 09:17:56)

#### Browser and version of browser, operating system running browser
Brave Version 1.10.97 Chromium: 83.0.4103.116 (Official Build) (64-bit)
MX Linux 19.2

#### Link to octoprint.log
https://gist.github.com/jbbandos/552f5aff8ba021022a513e06d807ee2f

#### Link to contents of terminal tab or serial.log
https://gist.github.com/jbbandos/4a6119ab4f979bebe752645ecba82b94

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jbbandos, 2020-07-08:
As I wrote above, I enabled HOST_ACTION_COMMANDS and HOST_PROMPT_SUPPORT in marlin. My issue seems to be the same as @AndKe 's one, so I didn't open a new ticket.
Unfortunately, I also think that this might also be a marlin issue, and not just an octoprint issue. The printer seems to start ignoring commands while paused.

jneilliii, 2020-07-08:
Just enabling host prompt support isn't enough I don't think. There were some action command variables removed from the default marlin source that broke it. It's stuff like action_on_pause AMD action_on_resume, or something like that.

AndKe, 2020-07-08:
mine was 
```
#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  //#define HOST_PROMPT_SUPPORT
#endif
```
it seems pointless to have the #define commented out by default, also, if all those ACTION_ON_* are required .. then the default config is just bad.

AndKe, 2020-07-08:
btw: ACTION_ON_PAUSE are configured by Marlin/src/inc/Conditionals_adv.h:78:   .. so maybe the defaults are not so bad at all - and adding them is not required?

jneilliii, 2020-07-08:
At some point it didn't work, maybe it got fixed since I was having the issue. So I guess the next question is, why is there no `action_on_filament_runout`?

jbbandos, 2020-07-09:
There is a marlin issue opened for the "other side" of this bug - [18565](https://github.com/MarlinFirmware/Marlin/issues/18565)
If the printer stops the serial connection, there isn't much that octoprint can do.

closed 2020-07-14 by GitIssueBot