OctoPrint/OctoPrint #3626: [RC Feedback] Feedback on the 1.4.1rc1 Release Candidate

foosel, 2020-07-06:
Please provide *general* feedback on your experience with the 1.4.1rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Unrelated third party issues

### Issues needing analysis

schnello, 2020-07-06:
System Klipper with TMC UART
1.4.0 > All is Fine
1.4.1 > Klipper is now unable to init the TMC via Uart.
fallback to 1.4.0 > All is fine again.

foosel, 2020-07-06:
@schnello in any case: 

> If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

;)

gege2b, 2020-07-06:
I'm doing a print right now, and except some plugin error (with octolapse at least - I need to do some check before saying it's a bug) and the high CPU usage, everything seems fine

jneilliii, 2020-07-06:
Cross referencing issue related to Print Time Genius and 1.4.1. I know this isn't OctoPrint's issue, but just in case someone comes across this.

https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/202

marknn3, 2020-07-07:
Unable to connect after update to 1.4.1rc1 on Raspberry Pi 2B with OctoPi 0.17.0.
Printer is Ender 3 with Marlin 2020-07-02 firmware on SKR Mini E3 V2.0.
Reverting back to 1.4.0 restored the connection.
```
2020-07-07 20:50:09,739 - Enabling serial logging
2020-07-07 20:50:13,533 - Changing monitoring state from "Offline" to "Detecting serial connection"
2020-07-07 20:50:13,642 - Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
2020-07-07 20:50:13,655 - Connecting to port /dev/ttyACM0, baudrate 115200
2020-07-07 20:50:18,738 - Trying port /dev/ttyACM0, baudrate 115200
2020-07-07 20:50:18,770 - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2020-07-07 20:50:18,801 - Send: M112
2020-07-07 20:50:18,804 - Send: N1 M112*32
2020-07-07 20:50:18,807 - Send: N2 M104 T0 S0*35
2020-07-07 20:50:18,810 - Send: N3 M140 S0*102
2020-07-07 20:50:21,405 - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2020-07-07 20:50:21,439 - Connected to: None, starting monitor
2020-07-07 20:50:21,442 - Connection closed, closing down monitor
```
Note that M112 ***is*** send to the printer, but this does shutdown the printer and is then unable to respond to any commands.

foosel, 2020-07-07:
Sounds exactly like #3627

ogsv, 2020-07-07:
@marknn3 Can you confirm if you are able to connect when manually setting the connection parameters? If yes, it's #3627 , but if not, we might be two ppl with the same issue...

marknn3, 2020-07-07:
@ogsv Ah I see. When I change Serial Port from AUTO to /dev/ttyACM0 then the connection succeeds! Auto Baudrate works correctly.
So indeed, it's #3627.

foosel, 2020-07-08:
1.4.1rc2 has just been released, new feedback ticket is here: #3638

closed 2020-07-08 by foosel

marknn3, 2020-07-08:
The connection problem is now solved for me in 1.4.1rc2 with Auto Serial Port.
Notice that in 1.4.1rc1 a M112 was successfully sent to the printer during connect and the printer did shutdown. This might be an undesirable side effect for not able to connect.
When I switch off M112 in Settings > Serial Connection > Behaviour, the M112 is not sent but the connection still fails.