OctoPrint/OctoPrint #3627: [1.4.1rc1] - Printer not detected when starting connection.

zeroflow, 2020-07-06:
#### What were you doing?

Update to 1.4.1 RC1.
Then tried connecting to my Prusa Mk3s.

1. Update to 1.4.1 RC1
2. Click connect

#### What did you expect to happen?

Connection to printer established

#### What happened instead?

Printer restarted (as expected when connecting) but Octoprint reported a failure to connect ("Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)")

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.4.1rc1 Python 2.7.13 OctoPi 0.15.1

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

Prusa Mk3S, Original Prusa 3.9.0

#### Browser and version of browser, operating system running browser

Google Chrome Version 83.0.4103.116 (Official Build) (64-bit)
Windows 10 latest

#### Link to octoprint.log

https://gist.github.com/zeroflow/457f83797e9cd37e50639dc67aea76ca

#### Link to contents of terminal tab or serial.log

https://gist.github.com/zeroflow/457f83797e9cd37e50639dc67aea76ca
```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
Connecting to port /dev/ttyACM0, baudrate 115200
Trying port /dev/ttyACM0, baudrate 115200
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Send: N3 M104 T1 S0*35
Send: N4 M104 T2 S0*39
Send: N5 M104 T3 S0*39
Send: N6 M104 T4 S0*35
Send: N7 M140 S0*98
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
Connected to: None, starting monitor
Connection closed, closing down monitor
```

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

No screenshots necessary.

I have read the FAQ.

zeroflow, 2020-07-06:
UPDATE: By manually selecting the Port & Baudrate, OctoPrint can connect to the printer.

foosel, 2020-07-06:
Seems to only happen when the port is set to AUTO but the baudrate isn't. I can reproduce it that way against the Virtual printer. Setting both to AUTO makes it work, as does setting both to a fixed value.

foosel, 2020-07-06:
Fix ready for 1.4.1rc2

DodgeDeBoulet, 2020-07-07:
I ran into this issue this morning with 1.4.1rc1 and even manual selection of the device and baud rate did not resolve it. I was able to connect with **sudo screen /dev/ttyACM0** on the Pi and Pronterface from a different system. Serial log is posted on the OctoPrint community forum, but it didn't seem to show anything enlightening.

I'm back on 1.4.0 GA now and everything is working again. I can get other files from the 1.4.1rc1 installation if needed, but the SD card it was on is apparently no longer bootable ... I have it mounted via the SDcard slot on my Ubuntu laptop though and can fetch anything needed for forensics.

ogsv, 2020-07-07:
I seem to be unable to connect even when setting both to AUTO or both to manual values:
`Connecting to port /dev/ttyACM0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x68014a10, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"`

As others, I can connect to the printer from a shell on the RPi. Only troubleshooting steps I've tried yet is rebooting the Pi and the controller (MKS Rumba32 running some random build of Marlin 2.0)

foosel, 2020-07-08:
@DodgeDeBoulet Based on the [logs](https://community.octoprint.org/t/cant-connect-to-the-printer-after-upgrading-to-1-4-0/16934/6?u=foosel) all I can see is that it opens the serial port just fine, tries to perform handshakes with the printer and that never replies. That part of the code was not touched between 1.4.0 and 1.4.1, so I don't understand why 1.4.0 works and 1.4.1 doesn't for you, but in any case it's certainly a different issue than what is described in this ticket, which definitely involves the rewritten auto detection. 

There's another issues in 1.4.1rc1 that causes ridiculously high CPU load that might have nasty side effects, so I suggest you see if you *still* can't connect once 1.4.1rc2 hits later today (I'll start on final testing and packaging right after I hit send on this) - if yes, please open a *proper bug report with a fully filled out ticket template and all requested logs*, if no, hooray.

foosel, 2020-07-08:
I just pushed 1.4.1rc2, but sadly this turns out to not be fixed after all, as I observed when I updated one of my printers. Tests against different hardware two days again were positive. It appears to be a timing issue with the new auto detection code that needs some further looking into, if the printer needs a longer time to boot before it responds to the `M110` probes the detection will fail. Also no error handling routine should be triggered here as it currently is.

foosel, 2020-07-08:
Next fix, ready for 1.4.1rc3. Hopefully this time it works in all cases. I've changed the timeout handling to work around long bootup cycles. Extra long ones might still cause issues, however they will also during non-auto-connect and we need to draw the line somewhere. There's timeout settings to work around that on an individual case-by-case base.

foosel, 2020-07-15:
1.4.1rc3 has been released

closed 2020-07-15 by foosel

Ricardo-Miguel-Caldeira, 2020-08-01:
This is still happening with 1.4.rc4 although selecting the correct port and baudrate will manage to work. 

`2020-08-01 05:43:15,155 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2020-08-01 05:43:15,159 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 115200
2020-08-01 05:43:15,192 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2020-08-01 05:43:15,197 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-08-01 05:43:53,040 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-08-01 05:44:03,060 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Offline"
2020-08-01 05:44:03,071 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-08-01 05:44:29,061 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2020-08-01 05:44:29,068 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 230400
2020-08-01 05:44:29,119 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2020-08-01 05:44:29,147 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-08-01 05:44:29,149 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
`

foosel, 2020-08-01:
Serial.log, printer model and firmware version please.

Prithul0218, 2020-08-13:
I'm also getting similar kind of problems. Haven't been able to get octoprint running for 7 hours now

jneilliii, 2020-08-13:
Would be great to get the same requested information from you @Prithul0218, enable serial logging in settings, restart OctoPrint, Provide logs, printer model and firmware version.

jneilliii, 2020-08-13:
and probably in an open issue, rather than a closed one.

jjl40, 2021-01-07:
I am having this problem all of a sudden. I have been using 1.4.2 for many months. without issue. Now I am getting this error "Error: No more candidates to test, and no working port/baudrate combination detected.". I upgraded to 1.5.2 and that didn't help. I restored my system from a backup and it started working again for about 1 day. Now I am getting the same error again. The only thing I changed was added a BigTreeTech TFT35. I don't think that would cause the problem as it isn't attached to the pi. It is just emulating the reprap discount full graphics display.
It's agrivating because even though I have a backup it still takes a LOT of time to get back up and running.

jneilliii, 2021-01-08:
> The only thing I changed was added a BigTreeTech TFT35

I think these act as a host controller, which could compete for the serial connection. You might try unchecking `Request exclusive access to the serial port` in the serial connection settings.

jjl40, 2021-01-08:
Thanks for the reply. Does this apply if you only use 12864 screen mode? 
I only have the ribbon cables connected. I haven’t figured out how to set up the serial port on my RAMBO v 1.3. 
I’ll try your suggestion tomorrow. I’ve enjoyed all of this I can stand today:)
John Lytle

DodgeDeBoulet, 2021-01-08:
It only acts as a host controller if you *a)* Also have it plugged into the
TFT (serial) port on the main board and *b*) Have it defined in Marlin as a
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. If you're only ever going to
use it in 12864 emulation mode, define it as a CR10_STOCKDISPLAY.

I think 😉

jjl40, 2021-01-08:
Thanks Dodge 

I’ll give that a try. 
That’s a great screen name!

John Lytle

jjl40, 2021-01-08:
Thanks for the suggestion but that didn't work. I won't have time to work on this for a couple weeks. I will let you know when I find a solution.

John Lytle

commit 46203fd4 (2020-07-06): 🐛 Fix auto detect with auto port and set baudrate

Closes #3627