OctoPrint/OctoPrint #3629: [1.4.1rc1] Logging error from tracking plugin on shutdown

cp2004, 2020-07-06:
#### What were you doing?

Running OctoPrint 1.4.1rc1 from the command line (Or Pycharm debugger, happens on both)
When I go to shutdown, (`ctrl-c` or the stop button) there is a logging error in the tracking plugin.

#### What did you expect to happen?
Exit cleanly
#### What happened instead?
Have pasted the error here, as it is a logging error it's not in the log!
```--- Logging error ---
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/logging/handlers.py", line 20, in emit
    self._executor.submit(self._emit, record)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 151, in submit
    raise RuntimeError('cannot schedule new futures after shutdown')
RuntimeError: cannot schedule new futures after shutdown
Call stack:
  File "/usr/lib/python3.7/threading.py", line 885, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/pi/oprint/lib/python3.7/site-packages/sentry_sdk/integrations/threading.py", line 67, in run
    return old_run_func(self, *a, **kw)
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 80, in _worker
    work_item.run()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking/__init__.py", line 425, in _do_track
    self._logger.info("Sent tracking event {}, payload: {!r}".format(event, kwargs))
  File "/usr/lib/python3.7/logging/__init__.py", line 1383, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python3.7/logging/__init__.py", line 1519, in _log
    self.handle(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 1529, in handle
    self.callHandlers(record)
  File "/home/pi/oprint/lib/python3.7/site-packages/sentry_sdk/integrations/logging.py", line 83, in sentry_patched_callhandlers
    return old_callhandlers(self, record)
  File "/usr/lib/python3.7/logging/__init__.py", line 1591, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 905, in handle
    self.emit(record)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/logging/handlers.py", line 22, in emit
    self.handleError(record)
Message: 'Sent tracking event shutdown, payload: {}'
Arguments: ()
--- Logging error ---
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/logging/handlers.py", line 20, in emit
    self._executor.submit(self._emit, record)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 151, in submit
    raise RuntimeError('cannot schedule new futures after shutdown')
RuntimeError: cannot schedule new futures after shutdown
Call stack:
  File "/usr/lib/python3.7/threading.py", line 885, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/pi/oprint/lib/python3.7/site-packages/sentry_sdk/integrations/threading.py", line 67, in run
    return old_run_func(self, *a, **kw)
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 80, in _worker
    work_item.run()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking/__init__.py", line 425, in _do_track
    self._logger.info("Sent tracking event {}, payload: {!r}".format(event, kwargs))
  File "/usr/lib/python3.7/logging/__init__.py", line 1383, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python3.7/logging/__init__.py", line 1519, in _log
    self.handle(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 1529, in handle
    self.callHandlers(record)
  File "/home/pi/oprint/lib/python3.7/site-packages/sentry_sdk/integrations/logging.py", line 83, in sentry_patched_callhandlers
    return old_callhandlers(self, record)
  File "/usr/lib/python3.7/logging/__init__.py", line 1591, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 905, in handle
    self.emit(record)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/logging/handlers.py", line 22, in emit
    self.handleError(record)
Message: 'Sent tracking event shutdown, payload: {}'
Arguments: ()
```
#### Did the same happen when running OctoPrint in safe mode?
Yes, tested in both

#### Version of OctoPrint
1.4.1rc1

#### Operating System running OctoPrint
OctoPi 0.17 on a local checkout (of rc/maintenance)

#### Printer model & used firmware incl. version
N/A (Not connected to printer as this is my dev pi)

#### Browser and version of browser, operating system running browser
N/A

#### Link to octoprint.log
Error does not show in the logs (As it is logging error)
My test which is in the gist goes Python 2(No error), Python 3, Python 3 safe mode.)

https://gist.github.com/cp2004/99d65bdbe9811b7f4ac54dac1148e4ba

#### Link to contents of terminal tab or serial.log
N/A (Not connected)

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

cp2004, 2020-07-06:
One more thing, If I disable anonymous usage tracking this no longer happens

foosel, 2020-07-06:
Does the same happen under 1.4.0?

cp2004, 2020-07-06:
@foosel Just tested it, and no it doesn't.

foosel, 2020-07-06:
Possibly something to do with the sentry sdk update then. Meh.

cp2004, 2020-07-06:
I think it was something fairly recent, as I was running the maintenance branch for developing the Py3 upgrade script and didn't notice it then. Can't pinpoint when though, but possibly between 2 weeks ago (Running maintenance to test upgrade) and 02/07 which was when I set up the new environment to start plugin development.

foosel, 2020-07-06:
Fix ready for 1.4.1rc2. A bit puzzled though that it didn't reproduce in 1.4.0, the issue was in the code for a long time now.

cp2004, 2020-07-06:
Cool, will let you know if I run into anything else!

cp2004, 2020-07-06:
Can't reproduce this issue now, closing

commit fa148a8b (2020-07-06): 🐛 Fix async issue with logging on shutdown

Asynchronous components might be active after on_shutdown, and
AsyncLogHandlerMixin should be able to cope with logs generated by
them.

Closes #3629
closed 2020-07-06 by cp2004