OctoPrint/OctoPrint #3630: [1.4.1rc1] Single File Plugin Install Fails in Python 2

jneilliii, 2020-07-06:
#### What were you doing?

Testing the new single file plugin feature by uploading a py file directly in plugin manager

#### What did you expect to happen?

Plugin to install

#### What happened instead?

Error message

#### Did the same happen when running OctoPrint in safe mode?

n/a

#### Version of OctoPrint

1.4.1rc1

#### Operating System running OctoPrint

OctoPi 0.18

#### Printer model & used firmware incl. version

n/a

#### Browser and version of browser, operating system running browser

Chrome

#### Link to octoprint.log

relevant error from log.
```
2020-07-06 09:56:09,876 - octoprint - ERROR - Exception on /plugin/pluginmanager/upload_file [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1311, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/access/permissions.py", line 174, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 285, in upload_file
    force="force" in flask.request.values and flask.request.values["force"] in valid_boolean_trues)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 465, in command_install
    if self._is_archive(path):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 432, in _is_archive
    kind = filetype.guess(path)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/filetype/filetype.py", line 28, in guess
    return match(obj) if obj else None
  File "/home/pi/oprint/local/lib/python2.7/site-packages/filetype/match.py", line 28, in match
    buf = get_bytes(obj)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/filetype/utils.py", line 72, in get_bytes
    raise TypeError('Unsupported type as file input: %s' % kind)
TypeError: Unsupported type as file input: <type 'unicode'>
```

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s)/video(s) showing the problem:

n/a

I have read the FAQ.

foosel, 2020-07-06:
Fix ready for 1.4.1rc2

foosel, 2020-07-08:
1.4.1rc2 has been released.

commit f1cc40a8 (2020-07-06): 🐛 PMGR: Fix single file plugins under Py2.7

filetype dependency requires paths as str, so we need to make sure
we provide what will be identified as a native str type under both
py2 and 3.

Closes #3630
closed 2020-07-08 by foosel