OctoPrint/OctoPrint #3631: Octoprint does not warn users that it cannot determine filament status on certain printers

alandsidel, 2020-07-06:
Octoprint does not warn you when you are using a printer that is known to not send filament status out of the serial port.  It should.

There was no warning that the built in filament sensor would not work with my CR-10s pro when I started a long (~18hr) print, in which I intentionally wanted to run out an old spool and add in a new one, that the sensor was not going to work and the print was going to continue on level by level doing nothing at all.

Now I have no idea where it stopped physically, or even what layer it was on, so I can't imagine how I can salvage the print.

This is extremely user unfriendly behavior.  I know that ultimately it's Creality's fault that the filament status isn't sent out of the serial port, but if the Octoprint developers and user community at large know that this is the case (and they do -- as a search for "octoprint cr-10s pro filament" indicates), they should at the very least throw up some warning in the UI about this behavior before starting a print, so you know to not use octoprint in this case and just go back to using the SD.

Sorry for my tone, but this is an easy fix to a known problem, so the fact that I had to just run into it without warning is extremely frustrating.

fieldOfView, 2020-07-06:
OctoPrint does not do anything with filament sensors. There are several plugins for OctoPrint that do. You must be using one (or more) of them. Your issues therefore is with that (those) plugin(s), not with OctoPrint. This issue queue is only for bugs in OctoPrint itself, not for plugins for OctoPrint. Please open an issue at the issue queue for the plugin(s) you use.

closed 2020-07-06 by fieldOfView

alandsidel, 2020-07-06:
> OctoPrint does not do anything with filament sensors. There are several plugins for OctoPrint that do. You must be using one (or more) of them. Your issues therefore is with that (those) plugin(s), [...]

I know it doesn't work with this printer.

Now.

Did you read my issue before closing it?

foosel, 2020-07-06:
So you know it doesn't work with your printer, know that's the fault of Creality, but somehow it should fall to OctoPrint to work around all that yet again? Have you opened a ticket with an equal tone at the vendor who *sold* you a product that doesn't adhere to established industry norms as well? 

I might have been inclined to check whether it was maybe somehow possible to identify the firmware in question (which - as a side note - can be *incredibly* tricky) and add a warning to the bundled FirmwareCheck plugin, but frankly, after your entitled attitude on display here I'm not very much motivated to do so anymore. If you want something implemented to make your life easier, asking nicely is usually the better way than storming in and making demands.

fieldOfView, 2020-07-06:
I did read your issue, but it was not clear to me if you were already using a plugin to monitor a filament sensor.

3d printers can have many different sensors and features. It is not the aim for OctoPrint to support all features of all printers. It is NOT - as you suggest - an "easy fix" to support all features of all printers (or warn that specific features of all printers may not be supported). That is what plugins are for. Plugins can make it easy to support certain features that are specific for a printer.

alandsidel, 2020-07-06:
Again, I think you didn't actually read it @fieldOfView 

I am not asking for sensor support, not did I say that adding it would be an easy fix.  I'm a developer.  I know it wouldn't be.

What  I asked for, and said would be easy, would be a **warning** before starting a print that the sensor isn't going to be used, and to not use octoprint if there's a chance the filament will run out -- or to install a plugin -- also an option _that I did not even know was an option_ about until this issue.  I hope this better explains the problem.

It's not that I want filament sensor support.  It's not that I need help installing or finding a plugin.  It's that _I didn't even know this was going to be a problem until the filament ran out and my print was ruined._

foosel, 2020-07-06:
I'm going to lock this now as I don't feel like it's leading anywhere.

@mrseeker created a (friendly) feature request related to that which I moved to the bundled Firmware Check plugin. If at all, core implementation will happen there, once all involved questions and challenges are resolved.