OctoPrint/OctoPrint #3633: [1.4.1rc1] High CPU load

hamster65, 2020-07-06:
Hi foosel,

I switched from German to English and to manual port settings. Fine so far, but, even in safe mode, I see this:

 PID USER      PR  NI    VIRT    RES    SHR S  **%CPU**  %MEM     TIME+ COMMAND
 1569 pi        20   0  257288  64428   8640 S **106.9**   7.2   6:45.44 octoprint
  647 pi        20   0   25904   5156   4260 S  **68.3**   0.6  10:12.74 mjpg_streamer
 1141 root      20   0    8180   2984   2368 R   0.7   0.3   0:05.86 top

Not sure if mjpg_streamer was that high before the update, might be. But octoprint certainly was not.

jneilliii, 2020-07-06:
Have you restarted your pi and tried running OctoPrint in safe mode, just to be sure it's not plugin related or some artifact post upgrade?

hamster65, 2020-07-06:
I rebooted several times since the upgrade but not after switching to safe mode.

I think is has got something to do with the printer connection: 
When I click disconnect, octoprint immediately reconnects!
I unplugged the usb connection to the printer -> load now 5%.

foosel, 2020-07-06:
Please fill out the full ticket template.

hamster65, 2020-07-06:
What were you doing?
Upgraded to 1.4.1rc1

#### What did you expect to happen?
cpu load around 5-10 %

#### What happened instead?
cpu load > 100 %, slow start, slow response from web interface

#### Did the same happen when running OctoPrint in safe mode?
yes

#### Version of OctoPrint
1.4.1rc1

#### Operating System running OctoPrint
4.19.118-v7+
OctoPi version    : 0.15.1
Raspian buster

#### Printer model & used firmware incl. version
corexy Marlin 2.0 bugfix

#### Browser and version of browser, operating system running browser
Firefox 78.01, Linux Debian sid

#### Link to octoprint.log
https://pastebin.com/x8H2gzKv

#### Link to contents of terminal tab or serial.log
https://pastebin.com/La2d3E3i

#### Link to contents of Javascript console in the browser
https://pastebin.com/M0dTE97C

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

Reperiel, 2020-07-06:
I have the same problem. For me it's related to the printer connection. Connected -> CPU > 100%, disconnected -> CPU < 10%.

foosel, 2020-07-06:
Fix ready for 1.4.1rc2

foosel, 2020-07-08:
1.4.1rc2 has been released.

commit 20a3700a (2020-07-06): 🐛 Fix busy loop

Closes #3633
closed 2020-07-08 by foosel