OctoPrint/OctoPrint #3634: [1.4.1rc1] Klipper UART connection failed

schnello, 2020-07-06:
System: Ubuntu 20
Octoprint: 1.4.1 RC1 (Pyhton 3 env)

If i install Octoprint 1.4.1 RC1, Klipper cant connect to the TMC via UART.
If i fall back to Octoprint 1.4.0 all works as expected (tested this now two times).

 
[Log_TMC.txt](https://github.com/OctoPrint/OctoPrint/files/4880327/Log_TMC.txt)

foosel, 2020-07-06:
I'm hoping to need more info here in which way that can even be caused by OctoPrint. From the looks of the log, OctoPrint connects to the serial port on both cases. It doesn't know anything about TMC initialization or such, that's all on Klipper. I'd need to know in what way connecting to the serial port could influence that initialization routine.

The auto detection routine was changed such that OctoPrint will now try to do a handshake acrid possible port/baudrate candidates until it finds something that matches. However in the log out doesn't even look like it's doing detection at all.

@KevinOConnor would you happen to be able to help shed some light on this by any chance? I don't even know how to begin to debug this.

Also @schnello, please create a `serial.log` of both scenarios. Log excerpts won't necessarily tell the whole story and the more information here the better.

schnello, 2020-07-06:
@foosel 
I guess also a serial log would not help in this case. A serial log shows only the serial connection to the Klipper host. And this connection is "good".  Would be great if a other Klipper User could also confirm this error.

In the meantime i try to create a log but its very hard to start Octoprint and Klipper at the same time. See please also the pic.

[image]

After stopping klipper:
[image]

I will keep trying.

jneilliii, 2020-07-06:
The High CPU load was handled in another ticket and already has a fix for the next release candidate.

foosel, 2020-07-06:
Could possibly be a side effect of the high CPU load issue. Fix is pushed to `staging/maintenance`. 

> I guess also a serial log would not help in this case. A serial log shows only the serial connection to the Klipper host.

It could still hint at potential issues (so, please, always share it when asked or when reporting a bug), but my earlier point was also that that connection is the only thing that OctoPrint core even understands to handle, it has no idea of TMC initializations and what else might be necessary.

schnello, 2020-07-06:
I installed now the latest staging/maintenance commit and all is working fine now. Looks like youre right with "Could possibly be a side effect of the high CPU load issue."

Send: DUMP_TMC stepper=stepper_x
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 00081913 IHOLD=19 IRUN=25 IHOLDDELAY=8
Recv: // TPWMTHRS:   0000003e TPWMTHRS=62
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      00000065 IFCNT=101
Recv: // OTP_READ:   0000000d OTP_FCLKTRIM=13
Recv: // IOIN@TMC220x: 20000341 ENN=1 PDN_UART=1 SEL_A=1(TMC220x) DIR=1 VERSION=0x20
Recv: // FACTORY_CONF: 0000000d FCLKTRIM=13
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      00000078 MSCNT=120
Recv: // MSCURACT:   00b600a6 CUR_A=166 CUR_B=182
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c0130000 CS_ACTUAL=19 stealth=1 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  00000024 PWM_SCALE_SUM=36
Recv: // PWM_AUTO:   0016003a PWM_OFS_AUTO=58 PWM_GRAD_AUTO=22
Recv: ok
Send: M105
Recv: ok B:29.1 /0.0 T0:25.4 /0.0

closed 2020-07-06 by schnello

foosel, 2020-07-06:
Perfect, thanks for testing and reporting back! 👍