OctoPrint/OctoPrint #3638: [RC Feedback] Feedback on the 1.4.1rc2 Release Candidate

foosel, 2020-07-08:
Please provide *general* feedback on your experience with the 1.4.1rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Unrelated third party issues

### Issues needing analysis

marknn3, 2020-07-08:
The connection problem is now solved for me in 1.4.1rc2 with Auto Serial Port.
Notice that in 1.4.1rc1 a M112 was successfully sent to the printer during connect and the printer did shutdown. This might be an undesirable side effect for not able to connect.
When I switch off M112 in Settings > Serial Connection > Behaviour, the M112 is not sent but the connection still fails.

foosel, 2020-07-08:
It is indeed, and as mentioned above #3627 is also not definitely fixed, I can still reproduce it, but only against a MK3 as its boot up takes long enough to trigger timeouts. Looking into all of that at the moment.

schnello, 2020-07-09:
Finished my 10 h print with 1.4.1 RC2 with Klipper as FW without any problems.

b-morgan, 2020-07-09:
Installed successfully. The plugin [Simple Emergency Stop](https://github.com/Sebclem/OctoPrint-SimpleEmergencyStop) settings don't appear to be working correctly. An issue has been created for that plugin.

JohnOCFII, 2020-07-09:
All good so far:

- Upgraded (from 1.4.0 to 1.4.1rc1 to 1.4.1rc2) with no issues
- Uploaded new GCODE
- Printed GCODE on Prusa MK3S/wMMU2 in single extruder mode
- GCODE viewer worked well
- Terminal worked well
- Timelapse worked well
- Prusa MMU2 Select Filament plugin worked well
- OctoSlack plugin worked well
- Print Job History plugin worked well

Somewhere along the line the Telegram plugin seems to have been uninstalled.  I'll have to look into that one.  Could be a user issue. :)

jbjones27, 2020-07-09:
Hi Gina,  As always, thanks for the excellent work you do!
Upgraded from 1.4.0 -> 1.4.1rc1 -> 1.4.1rc2 with a few prints on both rc1 and rc2.
No issues at all.

I'm curious...  With a source code repository like Github, I am wondering how you get regressions on some of the releases.  I would think the repository version controls would limit that.  Just curious how the regressions happen.  Thanks!

gege2b, 2020-07-10:
Tested, high CPU load is gone  (I didn't got the other issues)

I still can't load the octolapse panel through (it just shows the spinner indefinitly) (just pinging  @FormerLurker here just in case)
I disabled every 3rd party plugin except octolapse to test and this changed nothing
The only thing I saw in the console windows is something about pannesl.js that can't be loaded. 
There is nothing in logs (neither octoprint nor octolapse ones)

I'm sorry for the lack of informations, I had prepared everything and forget to post it yesterday... and I don't have access to my instance right now, but I promise I'll post more infos later today

foosel, 2020-07-10:
> pannesl.js

That sounds entirely unfamiliar but is probably related. I think @FormerLurker might currently be AFK, I'll try to look into Octolapse issues myself as well just in case.

foosel, 2020-07-10:
@jbjones27 

> I'm curious... With a source code repository like Github, I am wondering how you get regressions on some of the releases. I would think the repository version controls would limit that. Just curious how the regressions happen. Thanks!

Regressions in this case are bugs introduced through code changes. When adding new features, improving existing ones or fixing bugs, it can happen that new issues get introduced in the process. That's what is referred to here as regressions. Simply rolling back to a prior revision obviously solves the regression but also gets rid of the change that introduced them in the first place, so whatever improvement or bug fix caused them, and that's obviously undesired, so when regressions are detected, they need to be solved while keeping the causing change in place (ideally at least, if a change turns out to be fundamentally broken it needs to go).

gege2b, 2020-07-10:
> 
> 
> > pannesl.js
> 
> That sounds entirely unfamiliar but is probably related. I think @FormerLurker might currently be AFK, I'll try to look into Octolapse issues myself as well just in case.

You're right xD !
To be sure it was OP related, I searched for "pannels.js" and found a bunch of results... and I just realized that I searched on github globally instead of just octoprint's repo... 
I think I need some rest

cp2004, 2020-07-10:
Been running 1.4.1rc2 for my plugin development, involving the virtual printer and lots of stop-starts:
- Tried single file plugins for reporting little bits of information to the logs, they're brilliant! 
- All seems to run smoothly, have not run into many issues
Only thing I would say (Not sure if it is specific to 1.4.1 - if I get time I'll try it later) is that if I press 'reload' on the web interface either before or while the preemptive cache is going after a restart, it takes a really long time. Will have to do some more investigation, but if I leave it, it completes in <1 sec, if not then it takes about a minute.

If you think this is worth further investigation I can create a separate ticket with logs etc.

cp2004, 2020-07-10:
Update on the above issue ^ ^

The preemptive cache seems slower on 1.4.1rc2, than it does on 1.4.0. It is much faster if I don't run it with PyCharm's debugger (Note: I'm connecting via SSH) connected, but still takes longer if I click 'reload' before it has cached the interface.

foosel, 2020-07-10:
I haven't touched the preemptive cache at all O_O Like, I haven't even been fiddling around with stuff in its neighbourhood. That it takes a bit longer under debugging is normal (everything is slower then). That it takes longer to reload if you reload while the cache is being built is something I've noticed in the past (so not new to 1.4.1*) and attributed to resource fighting and locking of the cache. But I'll take a look at that.

cp2004, 2020-07-10:
I mean it's not much of a difference (maybe 10-20secs Vs 1) when it's running normally, but I really noticed it when the debugger was connected.

b-morgan, 2020-07-10:
Just an FYI, the pannesl.js and this issue, https://github.com/FormerLurker/Octolapse/issues/567, may be related.

foosel, 2020-07-10:
So, the issues with Octolapse vs Simple Emergency Stop (and possibly other plugins too?) were caused by a Syntax Error triggered in the latter upon JS minification, which got enabled in 1.4.1. Tracked as #3640. I've just implemented a fix - Minification will only be enabled for core code and libs now, not for plugins. I would love to enable it for plugins, but if a minified plugin js file triggers a Syntax Error it nukes the whole third party plugin bundle - there's no way to capture one such error as far as I understand how JS works in the browser. So no minified third party JS files for the sake of error resilience.

b-morgan, 2020-07-10:
Can Minification optionally be turned on for plugins? It would seem like getting rid of the Syntax Errors at the source would be a good thing.

foosel, 2020-07-11:
I'll take a look. An additional devel setting defaulting to false will probably suffice.

ChrisHeerschap, 2020-07-13:
Installed rc1 when it came out, ran a couple prints with no issues, went to rc2 when it came out, been running that since, had the printers busy this weekend and no issues. Running on two raspis, a 4 and a 2, connected to a MK3S and a Prusa Mini, respectively.

ctgreybeard, 2020-07-13:
> Somewhere along the line the Telegram plugin seems to have been uninstalled.  I'll have to look into that one.  Could be a user issue. :)

I experienced the same problem. The Telegram plugin is now missing from the system.

JohnOCFII, 2020-07-13:
> > Somewhere along the line the Telegram plugin seems to have been uninstalled.  I'll have to look into that one.  Could be a user issue. :)
> 
> I experienced the same problem. The Telegram plugin is now missing from the system.
> 
> Reinstalling it “succeeded” but the plugin is still missing. Installation gave a strange message at completion:
> 
> [image]

I submitted a new issue on the OctoPrint-Telegram Plugin Github and linked to your entry @ctgreybeard 
New issue there is: https://github.com/fabianonline/OctoPrint-Telegram/issues/269#issue-656110763

foosel, 2020-07-13:
Can y'all share a log with me? There must be an indication in there why the plugin no longer shows up.

cp2004, 2020-07-13:
Don't use telegram, but thought I'd install it to see what happens.
Looks to me like a dependency issue, spotted this in the flask-babel repo. https://github.com/python-babel/flask-babel/issues/155 
Deprecated way of importing removed in 1.0.0, that OctoPrint has now updated to.

``` 2020-07-13 21:34:07,950 - octoprint.plugin.core - ERROR - Error loading plugin telegram
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 5, in <module>
    from flask.ext.babel import gettext
ImportError: No module named ext.babel
```

JohnOCFII, 2020-07-13:
> Can y'all share a log with me? There must be an indication in there why the plugin no longer shows up.

In this zip file are my current octoprint.log and the plugin_pluginmanager_console.log

[OctoPrintLogs.zip](https://github.com/OctoPrint/OctoPrint/files/4915034/OctoPrintLogs.zip)

foosel, 2020-07-15:
1.4.1rc3 has just been released, new feedback ticket is https://github.com/OctoPrint/OctoPrint/issues/3643

closed 2020-07-15 by foosel