OctoPrint/OctoPrint #3639: Dutch translation for OctoPrint

Tillema, 2020-07-09:
Title says all.

I translated PrusaSlicer into Dutch as well, so, already have some experience.
I'll do it for free, and will provide up-to-date translations on new versions.

jneilliii, 2020-07-09:
https://community.octoprint.org/knowledge-explorer?search=translations&topic=21144

foosel, 2020-07-09:
Hi @Tillema 👋 Thanks for the interest in translating OctoPrint. As explained at the link that @jneilliii gave you, please create a language pack. I do not want to bundle languages with OctoPrint that I cannot speak myself as I would not be able to guarantee stuff to stay up to date that way and hence create the language pack mechanism so that the community can contribute translations and keep them up to date independently of me.

closed 2020-07-09 by foosel