OctoPrint/OctoPrint #3640: [1.4.1rc2] JS minification can cause issues with third party plugins, nuking all of packed_plugins.js

foosel, 2020-07-10:
### Problem

New default of minifying JS assets from third party plugins can cause issues if that triggers Syntax Errors inside one or more third party plugins. As SyntaxErrors cannot be caught, any such issue will effectively nuke the whole `packed_plugins.js` bundle.

See also Sebclem/OctoPrint-SimpleEmergencyStop#14 and FormerLurker/Octolapse#567

### Solution

Disable minification again by default, at least for anything involving third party plugins, as it was prior to 1.4.1.

### Note

The old behaviour can be enforced again and thus the above problem avoided by setting 

``` yaml
devel:
  webassets:
    minify: false
```

in `config.yaml`.

FormerLurker, 2020-07-11:
@foosel, could you stuff the JS from each plugin into a separate file and minify each separately?

foosel, 2020-07-11:
I could, but that would defeat the whole point of bundling the assets together to avoid a flood of requests needed to load the page. And we really want to avoid that as it increases page load times (and server load) significantly.

foosel, 2020-07-15:
1.4.1rc3 has been released

commit 6bac39c1 (2020-07-10): 🥅 Disable minification only for third party plugins

Better solution than 7915e6c04e049d75971ad4b6f6e1bb3469195623.

Closes #3640
closed 2020-07-15 by foosel