OctoPrint/OctoPrint #3641: [Request] Stop processing plugins from reading the uploaded file

mrseeker, 2020-07-13:
**Is your feature request related to a problem? Please describe.**
I found out that when I am doing a print, and I uploaded a file to octoprint, my printer did a little hiccup due to buffer underrun because plugins were firing and erroring out, all while the printer was printing something heavy. The hiccup was small but noticeable. I was hoping that the file upload did not affect the printing itself.

**Describe the solution you'd like**
While in the state "PRINTING", disable/queue all python hooks related to octoprint.filemanager and derivates to prevent lagging from plugins firing off while uploading files.

**Describe alternatives you've considered**
Wait 12 hours for the print to finish, then upload the file.

**Additional context**
I had this problem while printing a rather heavy print with a lot of Gcode commands together with Arc-welder, and it seemed to stop printing as soon as the file was uploaded, and before it returned an error from ArcWelder that the print could not be processed.

cp2004, 2020-07-13:
Looking through the Arc Welder source, this is something it already does? 
I see this part here [`ArcWelderPlugin/__init__.py` - L573](https://github.com/FormerLurker/ArcWelderPlugin/blob/fe31ce3be16eaa9549ea775a419ca6e03119559e/octoprint_arc_welder/__init__.py#L573), seems to do what you wanted?

Did you see the message that the printer is currently printing? If so it was probably the upload that slowed the printer, I have seen that happen. From looking at the ArcWelder source there is no reason it should affect performance if the printer is printing.

foosel, 2020-07-13:
That's not really doable, at least not if the plugins hook into the Upload event (as arc welder does). That needs to be fired contractually when a file is uploaded. Not doing so would break plugins with a different purpose and even core functionality. 

OctoPrint can't determine whether a plugin will do something resource intensive or not in response to an upload event or some other hook. Only the plugin knows, so it's the plugin that needs to not do resource intensive tasks while printing. 

The only thing I can do here really is add yet another big fat disclaimer to the documentation about the importance of avoiding resource intensive tasks while printing, but that's all.

mrseeker, 2020-07-14:
> Looking through the Arc Welder source, this is something it already does?
> I see this part here [`ArcWelderPlugin/__init__.py` - [...]

From what I have seen, the following happened:
1. I uploaded a file
2. At 99%, the printer suddenly stopped working (either because octoprint was busy writing to disk, firing hooks, etc?)
3. The alert message popped up
4. Printer resumed printing...

It's not often I have seen this happening though, but it suddenly became obvious. The alternatives I am considering would be putting the active file in RAM or renice octoprint to give it more CPU cycles :)

foosel, 2020-07-14:
> putting the active file in RAM

That won't solve the issue. The problem isn't reading fast enough from the file, it's sending data fast enough to the printer when other stuff is hogging the CPU core (and thanks to CPython's GIL, that's one core unless the thing that's hogging CPU gets started in its own subprocess).

> renice octoprint

Sure, that can be done, but not from within OctoPrint. That would rather be a task to be done inside OctoPi/on manual install, as OctoPrint should normally not be run as root, which is however required to (re)nice from within Python:

``` 
>>> import psutil
>>> p = psutil.Process()
>>> p
psutil.Process(pid=9020, name='python.exe', started='10:29:32')
>>> p.nice()
0
>>> p.nice(-10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/oprint/local/lib/python2.7/site-packages/psutil/__init__.py", line 737, in nice
    self._proc.nice_set(value)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/psutil/_pslinux.py", line 1518, in wrapper
    raise AccessDenied(self.pid, self._name)
psutil.AccessDenied: psutil.AccessDenied (pid=13530)
```

Higher process priority would only be a stopgap solution anyhow however. Plugins just really shouldn't do resource intensive stuff while printing. Alternatively I'd have to altogether block uploading during an ongoing print, and I'm really averse to that as it would nuke quite a lot of workflows out there :/

What confuses me in this specific case here is that as visible in ArcWelder's source, the plugin *does* take the state into account here, and you still saw a slow down. I fixed an issue with upload processing in 1.4.1 that might potentially explain such a behaviour, though not caused by ArcWelder as far as I can see. So... question round ;)

  1. What version of OctoPrint did you see this on?
  2. What other plugins do you have installed?
  3. And just because I got confused but this:

      > The alert message popped up

      Which one? That it could not process while printing or that it was processing?

mrseeker, 2020-07-14:
> That won't solve the issue. The problem isn't reading fast enough from the file, it's sending data fast enough to the printer when other stuff is hogging the CPU core (and thanks to CPython's GIL, [...]

For 1 and 2, this is the relevant part of the octoprint.log:

```
2020-07-12 17:03:20,336 - octoprint.server - INFO - OctoPrint 1.4.0
2020-07-12 17:03:20,340 - octoprint.plugin.core - INFO - 32 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Action Commands (0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_actioncommands
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Arc-Welder (0.1.0rc1.dev4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_arc_welder
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Consolidate Temp Control (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_consolidate_temp_control
|  Continuous Print (1.0.3) = /home/pi/oprint/local/lib/python2.7/site-packages/continuousprint
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
| !Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
|  Filament Sensor Matrix (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmatrix
| !Filament Sensor Reloaded (1.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload
| !Filament Sensors Revolutions (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentrevolutions
|  Firmware Updater (1.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/loginui
|  M73 Progress Plugin (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m73progress
|  MultiCam (0.2.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_multicam
|  Octolapse (0.4.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  PrintTimeGenius Plugin (2.2.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius
|  Simple Emergency Stop (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_simpleemergencystop
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Sticky Pad (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stickypad
|  Telegram Notifications (1.5.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
|  YouTube Live (0.5.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_youtubelive
```

For 3: Yes, I got an error message from ArcWelder that it could not process because it was currently printing. The behaviour though suggested that the ArcWelder plugin loaded the file before realizing it could not process.

I think I will wait for 1.4.1 to see if this solves the behaviour, thanks for your time :)

closed 2020-07-14 by mrseeker