OctoPrint/OctoPrint #3645: Webcam stream enabled by default

cyb0man, 2020-07-17:
#### What were you doing?
I have noticed that Octoprint by default will eat a lot of bandwidth, 500-700 kbps.
This was a normal install on a RPi.
No camera connected.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
I would suggest that camera stream is disabled by default, especially stupid if there is no camera connected.

#### What happened instead?
I disabled the feature and made reboot.
(Restart did not help!)

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

gdachs, 2020-07-17:
OctoPrint is not involved in the webcam streaming. That is done by mjpg-streamer that is not part of OctoPrint.

foosel, 2020-07-22:
Please fill out the *complete* ticket template and explain your issue a bit more. There's nothing to stream if there's no webcam connected, so either your bandwidth comes from something else (third party plugin? **Test in safe mode**) or there's a webcam connected after all - which btw only gets loaded if the control tab is active, unless you run some wonky browser that doesn't support that behaviour.

In short: It's completely unclear what your problem actually is and you didn't provide enough information to determine that.

closed 2020-08-05 by foosel