OctoPrint/OctoPrint #3646: [RC Feedback]  Real well-working update.  TouchUI bug blocks gui for 1.4.1rc3

moof-src, 2020-07-17:
I know this is a TouchUI problem and not really an Octoprint problem except that I would really like an easier way to prevent this problem from within OctoPrint rather than opening a terminal, logging in, restarting in safe mode, and removing the offending plugin.    feature request below:

#### What were you doing?

I have been using 1.4.1rc3 for a few days bringing up a new printer.  I've used it to upload firmware, send many g-code commands through the terminal.  I have been heavily using the temperature and control panels.  These features work great.  I shutdown my octopi earlier using the menu.  After plugging the octopi back in and connecting via a browser, it started up in TouchUI mode.  I did not click the TouchUI icon.   The usual interface to get back to normal mode from the TouchUI mode wasn't there.

#### What did you expect to happen?
I never expected TouchUI mode.   I could not get out.

feature request: I really wish OctoPrint had emergency key sequences to deactivate the printer (#1) and a different sequence to disable all plugins (#2). 

#### What happened instead?
I couldn't get out of TouchUI mode.
I tried uninstalling the plugin.  that worked.  After reinstalling the plugin, the browser started out in TouchUI mode again, and again with no way out.

#### Did the same happen when running OctoPrint in safe mode?
I used safe mode to remove the plugin.  

#### Version of OctoPrint

OctoPrint 1.4.1rc3 Python 2.7.16 

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Custom Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Chrome Version 83.0.4103.116

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

moof-src, 2020-07-17:
[image]

jneilliii, 2020-07-17:
This is probably caused by a conflict between TouchUI and another plugin. I've had to deal with those for several of my plugins that also reacted that way in TouchUI. Sometimes you can click the top right icon and get into the TouchUI settings to disable it again, and I assume their might be a URL parameter as well that might work to bypass it loading, but haven't really deep dived into that. I do agree that having a different means to start octoprint in safe mode from the UI without having full UI access would be a benefit though.

BillyBlaze, 2020-07-18:
TouchUI gets activated by:
1. your browser width was below 980
2. your browser has Touch capability
3. your browser user agent contains TouchUI (only happens through bootloader)
4. your url contains '#touch' or '?touch' in the url

After TouchUI is activated it will activate itself persistent by checking a localstorage key.
- You can disable the first two detections in the OctoPrint Settings under the TouchUI section.
- You can disable TouchUI within the TouchUI Settings (not to be confused with OctoPrint Settings)

If you cannot disable TouchUI through the TouchUI settings then TouchUI thinks you're running it through the bootloader. That should be a bug if you cannot see that and are not running the TouchUI bootloader.

A quick way to fix this is to remove the 'TouchUI' key/value in your localstorage. (can be done in the DevTools) and disable the detection that was the culprit.

moof-src, 2020-07-18:
> TouchUI gets activated by:
>
>    1. your browser width was below 980
>
> I opened my browser on a 4k monitor running macos.

>
>    1. your browser has Touch capability
>
> don't know

>
>    1. your browser user agent contains TouchUI (only happens through
>    bootloader)
>
> this is a desktop system.  what bootloader?

>
>    1. your url contains '#touch' or '?touch' in the url
>
> my URL didn't have #touch.  I connect tith the default url,
http://hostname.local/

>
> After TouchUI is activated it will activate itself persistent by checking
> a localstorage key.
>
>    - You can disable the first two detections in the OctoPrint Settings
>    under the TouchUI section.
>
> I removed TouchUI and all settings.  I restarted my Octopi.  I reinstalled
TouchUI.  I restarted my OctoPi.  When I connected, it started in TouchUI
mode.  Many of the normal items in TouchUI were not there, especially the
screen that lets me turn TouchUI off.

>
>    - You can disable TouchUI within the TouchUI Settings (not to be
>    confused with OctoPrint Settings)
>
> The TouchUI settings do not show up when this bug occurs.

cp2004, 2020-07-18:
Updated my issue on the TouchUI repo, as I too was unable to exit touchUI as the settings menu was missing. Had to use a different browser and disable the plugin

foosel, 2020-07-21:
> feature request: I really wish OctoPrint had emergency key sequences to deactivate the printer (#1) and a different sequence to disable all plugins (#2).

I've been meaning to add some kind of recovery page for a while now, so this will be accessible under `/recovery/` starting with 1.4.2:

[image]

Very basic page, providing access to the core system commands, allowing to cancel the current print job (if one is ongoing), disconnect from the printer (if connected), and create/restore backups (if the plugin hasn't been disabled and the underlying system supports restoring). 

Access is restricted to admin accounts obviously.

cp2004, 2020-07-21:
@moof-src Might want to take a look at my linked issue (above) for TouchUI to confirm the fix. I believe the fox has gone against TouchUI/maintenance, but there may be a release soon.

foosel, 2020-08-06:
Make that 1.4.3, 1.4.2 just got turned into a hotfix release.

foosel, 2020-08-10:
Actually make that 1.5.0 because I've decided to finally *fully* adhere to semver and maintenance releases will see a minor increase in the future instead of patch, patch will be used for hotfixes if needed.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel