OctoPrint/OctoPrint #3649: Long delay from "Pausing" to "Paused" state

muistiloppu, 2020-07-20:
My set up is Raspberry PI 3B+ with Octopi (OctoPrint 1.4.0) controlling printer through the USB. Printer is based on Marlin 2.0 running on atmega2560 MCU.
  Problem is when printer has been printing long period of time and I manually pause it or filament run out happens. It can have tens of minutes delay from "Pausing" to "Paused" state.
 If pausing happens more often on the Octopi, then delay is in seconds.
 It feels like Octopi has a data buffer, what needs to be emptied before it can pause the printer. Or is there some other settings for a pausing?
   I have only 50mm distance from the filament sensor to feeder.

Here is a snippets from the logging files:

"octoprint.log.2020-07-16":

2020-07-16 20:44:13,706 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2020-07-17 16:37:44,043 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
**~20hours of printing and ~45min from "Pausing" to "Paused"**
2020-07-17 17:32:33,236 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"

"serial.log.2020-06-13_18-47-36":

2020-06-12 23:51:45,722 - Changing monitoring state from "Printing" to "Pausing"
2020-06-12 23:51:47,510 - Changing monitoring state from "Pausing" to "Paused"
2020-06-12 23:52:19,577 - Changing monitoring state from "Paused" to "Resuming"
**~10min of printing and ~10sec from "Pausing" to "Paused"**
2020-06-13 00:02:13,595 - Changing monitoring state from "Printing" to "Pausing"
2020-06-13 00:02:23,675 - Changing monitoring state from "Pausing" to "Paused"
2020-06-13 00:13:23,359 - Changing monitoring state from "Paused" to "Resuming"
**~17hours of printing and ~12min from "Pausing" to "Paused"**
2020-06-13 17:08:24,102 - Changing monitoring state from "Printing" to "Pausing"
2020-06-13 17:20:12,855 - Changing monitoring state from "Pausing" to "Paused"
2020-06-13 17:20:42,803 - Changing monitoring state from "Paused" to "Resuming"
**~8min of printing and ~20sec from "Pausing" to "Paused"**
2020-06-13 17:29:07,357 - Changing monitoring state from "Printing" to "Pausing"
2020-06-13 17:29:28,237 - Changing monitoring state from "Pausing" to "Paused"
2020-06-13 17:29:44,307 - Changing monitoring state from "Paused" to "Resuming"
**~11min of printing and ~18sec from "Pausing" to "Paused"**
2020-06-13 17:41:11,440 - Changing monitoring state from "Printing" to "Pausing"
2020-06-13 17:41:29,743 - Changing monitoring state from "Pausing" to "Paused"
2020-06-13 17:41:52,265 - Changing monitoring state from "Paused" to "Resuming"
**~40sec of printing and ~1sec from "Pausing" to "Paused"**
2020-06-13 17:42:34,848 - Changing monitoring state from "Printing" to "Pausing"
2020-06-13 17:42:35,161 - Changing monitoring state from "Pausing" to "Paused"

Original files are here:
https://drive.google.com/drive/folders/1IuN2AXJmPsIlsDEW_a7OccO4upSScERy?usp=sharing

foosel, 2020-07-22:
For some reason it looks like there's a TON of lines in your sending queue when the pause command is received, and those are sent to the printer first before the pause is finalized. The question is why that is the case. It's even visible in the log for your 11min print.

Before I start looking into this, could you please test if you can reproduce this in safe mode as well? An 11min print paused late should suffice to figure that out, if you share the `serial.log` from that with me.

muistiloppu, 2020-07-23:
I was printing with safe mode and still Octoprint is having long delay with pausing.
Pausing was done by clicking "Pause" on Octoprint browser window.
Delay is longer with large size parts and slow printing speed. 
Small size part with lot of short tangents and higher feedrate can keep up better with a G-code feed from the Octoprint and I guess thats why this issue does not show up with a small size printers.

**Here is a summary of the test print:**
**Quick pausing test**
2020-07-22 11:41:43,149 - Changing monitoring state from "Printing" to "Pausing"
2020-07-22 11:41:45,335 - Changing monitoring state from "Pausing" to "Paused"

**Quick pausing test**
2020-07-22 11:45:11,908 - Changing monitoring state from "Printing" to "Pausing"
2020-07-22 11:45:14,259 - Changing monitoring state from "Pausing" to "Paused"

**Printing 100% feedrate 6hours**
2020-07-22 18:10:51,665 - Changing monitoring state from "Printing" to "Pausing"
2020-07-22 18:12:22,888 - Changing monitoring state from "Pausing" to "Paused"

**Printing 100% feedrate 16hours**
2020-07-23 10:31:15,964 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 10:35:10,725 - Changing monitoring state from "Pausing" to "Paused"

**Printing 50% feedrate 2hours**
2020-07-23 12:37:06,049 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 12:37:34,776 - Changing monitoring state from "Pausing" to "Paused"

**Printing 150% feedrate 2hours**
2020-07-23 14:38:13,316 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:38:25,267 - Changing monitoring state from "Pausing" to "Paused"

**Quick pausing test**
2020-07-23 14:41:25,811 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:41:46,035 - Changing monitoring state from "Pausing" to "Paused"

**Quick pausing test**
2020-07-23 14:42:33,032 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:42:40,411 - Changing monitoring state from "Pausing" to "Paused"

**Quick pausing test**
2020-07-23 14:43:29,957 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:43:40,451 - Changing monitoring state from "Pausing" to "Paused"

**Quick pausing test**
2020-07-23 14:44:02,920 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:44:08,538 - Changing monitoring state from "Pausing" to "Paused"

**Quick pausing test**
2020-07-23 14:45:23,236 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:45:25,784 - Changing monitoring state from "Pausing" to "Paused"

**Printing 50% feedrate 60sec**
2020-07-23 14:46:45,718 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:46:56,868 - Changing monitoring state from "Pausing" to "Paused"

**Printing 50% feedrate 60sec**
2020-07-23 14:48:17,475 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:48:28,769 - Changing monitoring state from "Pausing" to "Paused"

**Printing 150% feedrate 60sec**
2020-07-23 14:49:55,808 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:50:02,348 - Changing monitoring state from "Pausing" to "Paused"

**Printing 150% feedrate 60sec**
2020-07-23 14:51:17,791 - Changing monitoring state from "Printing" to "Pausing"
2020-07-23 14:51:26,868 - Changing monitoring state from "Pausing" to "Paused"

**Original files:**
https://drive.google.com/file/d/1D_v2BsV7qPnmweFqQRo9ahyEY_a6aIGb/view?usp=sharing

foosel, 2020-09-07:
I have not been able to reproduce this, but based on some source code analysis I'm pretty sure it's fixed now on the `maintenance` branch. If you give that a whirl to check if you can confirm that would be awesome.

muistiloppu, 2020-09-11:
Thank you foosel for fixing this bug!
I have been testing this fix now couple days  and it looks good. Printer is pausing now with out any delays.

closed 2020-09-11 by muistiloppu