OctoPrint/OctoPrint #3650: Free/Used Space is Blank on rc3. and Unable to successfully upload gcode.

dcodom, 2020-07-21:
What were you doing?
I am using OctoPrint rc3. I have ran into two issues....
I just upgraded from the latest stable release.

Since upgrading I no longer see what the drive space used or available is.
and
I just tried the drag and drop method along with the manual upload. And the screen never says that it is finished saving. If I refresh the browser, it shows the file in the list, but I am not sure it is all there.

Version of OctoPi
1.4.1rc3

Printer model & used firmware incl. version
Ender3 Pro
RaspPi Version 3b+

Screenshot(s)/video(s) showing the problem:
image
[image]

I rolled the software back to 1.4.0 and it is all working as it should.

foosel, 2020-07-21:
I need logs 😕

dcodom, 2020-07-21:
Do you want all log files or specific ones? Wished there was a plugin/feature via the select all option to zip up the selected files and download...lol    Will cleanup my download directory and download each one. Just waiting for current print job to finish and then I will send it/them. It should be completed in about an hour or so.

jneilliii, 2020-07-21:
Probably just need the latest octoprint.log.

dcodom, 2020-07-21:
[octoprint (2).log](https://github.com/OctoPrint/OctoPrint/files/4955794/octoprint.2.log)

foosel, 2020-07-21:
Don't see any issue in the log, might be a front end only issue. Can't reproduce it on my end though, so maybe a third party plugin is interfering. Next steps:

  * Testing if the issue persists in safe mode
  * Sharing your js error console during reproduction of the issue

In short, all the points from the issue template you ignored 😉

dcodom, 2020-07-21:
Understood. I will perform the update/upgrade again and update the issue
tracker with the results.

Thanks for looking into it.

--DuWayne

dcodom, 2020-07-21:
[FirstbootAfterUpgrade.log](https://github.com/OctoPrint/OctoPrint/files/4956264/FirstbootAfterUpgrade.log)

This is from the error console.

Findings so far.... If I go into safe mode, both issues do not exist. I will try to disable each 3rd party plugin to see if one of them are causing the issue. If none of them, then I will try to do the plugins that come with the distribution.

Thanks :-)

dcodom, 2020-07-21:
Bingo!!!! GcodeEditor (0.2.8) is what is causing the problem. I will let them know, and see if they can fix. I do not use that plugin much so I will leave it disabled.

dcodom, 2020-07-21:
I have opened an issue with the author of the plugin as seen in previous comment.

Thanks for all your help.

I dont know if I should go ahead and close this issue or leave open so I can post the resolution.

foosel, 2020-07-21:
Yay, glad we got that figured out! 🥳

I'll close this one as it's not actionable for me (third party plugin issue) but I'll still keep an eye out for any updates related to this 😊

closed 2020-07-21 by foosel