OctoPrint/OctoPrint #3651: Terminal line styles to indicate source [Request]

jeffeb3, 2020-07-22:
**Is your feature request related to a problem? Please describe.**
Nope. Everything is awesome!

**Describe the solution you'd like**
What if we separated the messages going to and coming from the controller by color, and maybe also by justification? 

Maybe it could look like this?:

```
                               G1 X0
ok
                             G1 Y100
ok
                                M851
Probe Offset X30.00 Y0.00 Z-0.52
ok
```
Colors, like blue/red might also be really nice to show the difference. I can see from inspecting the page that each row is it's own span, so it seems possible to do this. I do not understand where these spans are coming from, or how to get their source all the way to the view template.

Since some output, like the format from G29 T or M122 needs to be left justified and monospace, I think it makes sense to put Marlin's text on the left.

**Describe alternatives you've considered**
I really haven't thought about it for more than a few minutes... :grimacing:...Sorry. I suppose I could write my own plugin to do this, but I'm not sure if I can edit so much of the existing structure. I suppose I could make a new tab with this, but that would definitely be harder than doing it in the main terminal. Just separating the spans by color might be enough to really make a difference.

**Additional context**
This feature reminds me of a text message, so here is some additional context:

[image]

cp2004, 2020-07-22:
This would be a really cool plugin idea, I see it having custom colours etc. for oks, vs other things. If I didn't have two plugins already in the pipeline I would take it.... So I'll leave this one for now.

jneilliii, 2020-07-22:
So this would be a fairly easy plugin to create with just a little javascript to do the work...
```
$('span:contains("Recv:")').addClass('text-left');
$('span:contains("Send:")').addClass('text-right');
```

jneilliii, 2020-07-22:
@jeffeb3 here's you a plugin. It's not perfect as some of the commands will flicker plain text for a second, but is close to what you want. Probably just needs some tweaking to what's being monitored in order to make those little flashes go away.

[image]

You can install using by copying/pasting the URL below in plugin manager > Get More > ...from URL.

https://github.com/jneilliii/OctoPrint-TerminalMessaging/archive/master.zip

jneilliii, 2020-07-22:
Unfortunately, removing the `Send:` and `Recv:` is possible by over-riding some of the internals of OctoPrint, but monkey-patching is frowned upon.

jneilliii, 2020-07-23:
Never mind, took a javascript approach and was able to remove those. I just pushed an updated to the repository.

[image]

jeffeb3, 2020-07-23:
Oh! That's awesome! That looks so clear to me. I also have a little chuckle at the text messaging triangles. It is exactly what I asked for, for better or worse. :100: Open source points for you!

I apparently need to learn some css, because I had no idea you could do that much with it. I get the align right and the background color, but I did not realize you could do things like those triangles with css.

So, how do I campaign for this excellent plugin to be part of the main octoprint? :)

I think this is awesome, but it maybe went too far to reach the text messaging level, but is actually not as functional as it could be. I will make a PR in your repo so I can show you what I was thinking.

closed 2020-07-23 by jeffeb3

jneilliii, 2020-07-23:
I probably won't do anything to maintain this plugin and don't believe it should be integrated to core because IMHO it's overkill. Feel free to fork the plugin and take ownership and make whatever changes you want. Happy to give you a starting point to make it better. One thing that might be nice for octoprint core is to have those individual lines classed in a way that allows people to style them however they want, ie using Themeify, which is basically what I'm doing programmatically to the different types. This would eliminate the need of this specific plugin.

jeffeb3, 2020-07-23:
That makes sense. That wouldn't be able to get rid of the Recv/Send items though.