OctoPrint/OctoPrint #3652: 404 error accessing websocket when URL ends with /

cerebrate, 2020-07-23:
#### What were you doing?

When using the OctoPod iPhone app, where the URL to connect to the OctoPrint server is entered ending with a /, the OctoPod client shows a "Not refreshing" error, and the OctoPrint logs show repeated instances of:

`2020-07-22 21:31:39,837 - tornado.access - WARNING - 404 GET //sockjs/websocket (::ffff:172.16.1.9) 14.59ms`

With quick testing, I found that this occurs when the URL of the OctoPrint instance is entered with a trailing slash; i.e.,

http://octoprint.example.com/

fails, while

http://octoprint.example.com/

succeeds.

To reproduce:

- Configure OctoPod client to use a URL ending with a slash.
- Connect to OctoPrint and observe the problem.
- Remove the slash and reconnect.
- Observe the not-problem.

#### What did you expect to happen?

OctoPrint should not care whether the URL ends in a slash or not (as is, for example, the typical albeit not default behavior of web servers), working identically in either case.

#### What happened instead?

OctoPrint fails obscurely when the URL ends in a slash.

#### Did the same happen when running OctoPrint in safe mode?

Yes, problem is identical in safe mode.

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

OctoPi-0.17.0

#### Printer model & used firmware incl. version

Creality Ender 5 Plus, stock firmware

#### Browser and version of browser, operating system running browser

N/A (not accessed via browser)

#### Link to octoprint.log

https://gist.github.com/cerebrate/5085a0ab953e7513852e1d141cc78762

#### Link to contents of terminal tab or serial.log

Not available (client does not retrieve).

#### Link to contents of Javascript console in the browser

Not available (not in browser).

#### Screenshot(s)/video(s) showing the problem:

[image]

jneilliii, 2020-07-23:
Your failing and succeeding examples are exactly the same? Outside of that, I feel this is more of an issue with OctoPod rather than the core OctoPrint and I see you've already opened an issue there.

cerebrate, 2020-07-23:
Argh. Damn autocorrect. Fixed.

On the latter, at least by my understanding of the problem, this seems like it would occur with any OctoPrint API client that subscribed to push updates with a /-terminated URL? I can see the argument that the client should be responsible for cleaning that up first, certainly, but on the other hand, there's one OctoPrint and an arbitrary number of potential clients, so handling it on the OctoPrint side makes sense from that PoV.

(And I suspect API client devs are every bit as susceptible to assuming web norm as API client users.)

cp2004, 2020-07-23:
Have just tried this, and tornado/OctoPrint redirects urls like `<ip>:5000/api//version` to `<ip>:5000/api/version`, however it does not redirect when it's the first one doubled (Like `<ip>:5000//api/version`). Although in that case, it works anyway.
The only bit that breaks is the websocket if you try `//sockjs/websocket` you get 404 from what I can see. Maybe other endpoints too, but I haven't tried.

foosel, 2020-07-27:
If a client just blindly attaches the endpoint addresses with a leading / to the configured base URL without checking if it's already /-terminated, that's a bug in the client. The web socket address is specified as `/sockjs/websocket`, not `//sockjs/websocket`. I'm honestly not too keen on making OctoPrint ever more complex to compensate for bugs in third party software :/ Especially if that means (monkey) patching third party libraries that I use (here: sockjs and/or tornado) to do so. 

I'm honestly surprised the double slash is not causing issues on the flask backed API endpoints, and if at all I'd rather look into fixing that behaviour so that `//api` also causes a 404.

The issue discussed here however should really get fixed in OctoPod.

closed 2020-07-27 by foosel