OctoPrint/OctoPrint #3653: [1.4.1rc3] Aborting auto-connection makes the connection box crash

cp2004, 2020-07-23:
#### What were you doing?
Testing 1.4.1rc3 while trying to develop my VirtualPrinter Settings interface. Was connecting/disconnecting to test that the settings bindings worked as intended. On my (Windows) machine, I have 2 ports COM3 and VIRTUAL availible. Looking closer at the device it says that COM3 is Bluetooth serial.

OctoPrint then requires a restart for the interface to be usable again, reload doesn't help

1. Enable autodetection with a port available that is NOT one that OctoPrint can connect to
2. Press 'connect'
3. Wait for it to start testing the ports, then press 'disconnect'
4. Watch terminal for disconnect note

#### What did you expect to happen?
Connection to be closed, button would switch to 'connect' and be responsive to further input

#### What happened instead?
Hung in the connected state, as shown here:

[image]

#### Did the same happen when running OctoPrint in safe mode?
I have no plugins installed (Except my settings template), and it did happen in safe mode

#### Version of OctoPrint
OctoPrint 1.4.1rc3, on both Python 2 & 3

#### Operating System running OctoPrint
Windows 10

#### Printer model & used firmware incl. version
Virtual!

#### Browser and version of browser, operating system running browser
Chrome and (new) Edge

#### Link to octoprint.log
[octoprint-1.4.1rc3 issue.log](https://github.com/OctoPrint/OctoPrint/files/4968794/octoprint-1.4.1rc3.issue.log)

#### Link to contents of terminal tab or serial.log
Have added in a line in the serial log to show when I pressed the button
[serial-1.4.1rc3.log](https://github.com/OctoPrint/OctoPrint/files/4968797/serial-1.4.1rc3.log)

#### Link to contents of Javascript console in the browser

No errors, but here's the log
[browserlog-1.4.1rc3issue.log](https://github.com/OctoPrint/OctoPrint/files/4968817/browserlog-1.4.1rc3issue.log)

#### Screenshot(s)/video(s) showing the problem:
Here's a little video of me testing it. It is not the whole web interface, just the connection bit that becomes unresponsive.
[image]

I have read the FAQ.
Hopefully that's all useful, I am in the process of installing 1.4.0 to see what happens there but it's taking forever at the moment...

cp2004, 2020-07-23:
Can confirm this does not happen in 1.4.0, the connection process ends and I can restart it easily.

foosel, 2020-07-27:
I'm not 100% sure I was able to reproduce this, but I did observe something that looks very much like what you describe above and was able to fix it through several commits. No longer able to reproduce. At least one of the related problems was present in 1.4.0 and earlier as well, so it might just be coincidence that it now gets triggered more easily. Race conditions are nasty little monsters.

In any case, fix ready for 1.4.1rc4

cp2004, 2020-07-27:
@foosel Switched to `staging/maintenance`, and I can still reproduce. However it *feels* like it's closer to being fixed, as the messages come through quicker & is more responsive. If it is any help, the terminal shows `Connection closed, closing down monitor` 3 times, once when I press the button, then again once it has shown the errors of auto detecting & has closed send loop, then finally a few seconds later. 

The other interesting thing I noticed (And you can see in the video above) is that when I press 'disconnect' again any other time, the port changes from `AUTO` to  `VIRTUAL`, and a request is sent back to the server, server calls `on_event`  but then gets no further. On a successful disconnect there are other events and info about the printers firmware etc. returned. 

If there is anything you want me to run or have a look at, let me know and I can (hopefully) provide it. If it is specific to the serial config here, my windows machine has a port that always exists, COM3 (Standard Serial over Bluetooth Link) and the `VIRTUAL` printer port. Can try with the actual printer connected, as that's connected to the RPI.

foosel, 2020-07-27:
Hm... that's bad. I cannot reproduce it myself anymore, similar setup though. Windows machine with a non-printer serial port COM6 plus virtual.

Does it solve itself if you reload the page?

cp2004, 2020-07-27:
No, just tried. Doesn't seem to solve it

foosel, 2020-07-27:
In that case it's indeed something else than what I was observing there myself, as the issue I could reproduce was a loss of state messages from the backend. Hm.... that makes this very tricky. Is there any chance I could access the affected instance myself via browser?

cp2004, 2020-07-27:
@foosel Have sent a DM on discord for ways to get this sorted

foosel, 2020-07-27:
NOW it should really be fixed.

Had to simulate your COM port (thanks for access to your instance so I could check out behaviour) via the accidentally code snippet that I removed again in 57b085f, but then I could reproduce it and find a fix, which is ready for 1.4.1rc4

foosel, 2020-07-29:
1.4.1rc4 has just been released.

commit 0cfc0874 (2020-07-27): 🐛 Set closed state even if not yet open

And also ensure we only attempt serial auto detect if we are still in
the correct state for it.

Closes #3653
closed 2020-07-29 by foosel