OctoPrint/OctoPrint #3654: Telegram Plugin missing

shinok00, 2020-07-25:
After updating to Octoprint 1.4.1 RC2 the  installed OctoPrint-Telegram Plugin 1.5.0 is missing and not working anymore.
I tryed to reinstall it, but after restart its still missing.

[Archiv.zip](https://github.com/OctoPrint/OctoPrint/files/4976075/Archiv.zip)

JohnOCFII, 2020-07-25:
> After updating to Octoprint 1.4.1 RC2 the installed OctoPrint-Telegram Plugin 1.5.0 is missing and not working anymore.
> I tryed to reinstall it, but after restart its still missing.
> 
> [Archiv.zip](https://github.com/OctoPrint/OctoPrint/files/4976075/Archiv.zip)

It is a known issue.  It is not an OctoPrint issue, but something the Telegram plug-in author needs to correct.  First referenced here: https://github.com/OctoPrint/OctoPrint/issues/3638#issuecomment-657712892

The issue has been logged with the Telegram GitHub, but they need to take action to correct it.  https://github.com/fabianonline/OctoPrint-Telegram/issues/269#issue-656110763

foosel, 2020-07-26:
Yep, and I even shot the maintainers a PR.

closed 2020-07-26 by foosel