OctoPrint/OctoPrint #3655: [1.4.RC3] File List API with location broken

jneilliii, 2020-07-26:
#### What were you doing?

Testing a new android app for octoprint for a Discord user. When loading file list there is a 500 Internal Server error. After further investigation, seems to be broken altogether when using the URL http://<ip>/api/files/local?recursive=true

#### What did you expect to happen?

Recursive file list of all local files.

#### What happened instead?

500 Internal Server Error

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.1RC3

#### Operating System running OctoPrint

OctoPi 0.18

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 84.0.4147.89 (Official Build) (64-bit)

#### Link to octoprint.log
relevant error
```
2020-07-26 19:05:40,910 - octoprint - ERROR - Exception on /api/files/local [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1109, in decorated_function
    lm = lastmodified_factory()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/files.py", line 135, in <lambda>
    request.values.get("recursive", False)),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/files.py", line 66, in _create_lastmodified
    storage, path_in_storage = path.split("/", 1)
ValueError: need more than 1 value to unpack
2020-07-26 19:05:41,012 - tornado.access - ERROR - 500 GET /api/files/local?recursive=true (::ffff:192.168.0.119) 111.94ms
```
#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

jneilliii, 2020-07-26:
I've tested in 1.4.0 just now and this issue does not occur.

foosel, 2020-07-27:
Looks like a nasty side effect of the fix to #3422

foosel, 2020-07-27:
Fixed by the above commit, ready for 1.4.1rc4.

foosel, 2020-07-29:
1.4.1rc4 has just been released.

commit 0188745d (2020-07-27): 🐛 Fix 500 on /api/files/local

Closes #3655
closed 2020-07-29 by foosel