OctoPrint/OctoPrint #3657: Chrome Version 84.0.4147.89 crashes client side of webservice.

brbrindle, 2020-07-27:
#### What were you doing?
Octoprint 1.4.0 on Octopi 1.7 
Loading Octoprint client from Chrome Version 84.0.4147.89 (x64) on both windows and linux, results in browser hang if settings, or cancel buttons are clicked. 

Normal operations seems unaffected (upload, print, all tabs...)

Android Chrome of same version seems unaffected. Firefox 78.0.2 on both windows and linux seems unaffected.
Multiple computers confirmed same issue.

Attempting to inspect elements is unhelpful. Clearing cookies and browser cache did not help.

Octoprint service seems to be fine with all of this, continuing printing etc. top shows no unusual cpu activity
Attempted clean install 3 times thinking there was something else going on. I am 95% certain this is an Octoprint issue with Chrome.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-07-27:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * whether you have been able to reproduce this issue in safe mode too
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)
  * screenshots or maybe a video showing the problem

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

brbrindle, 2020-07-27:
safe mode seems to make no difference... 

JS log... show normal login, then after settings button clicked... no updates.

octoprint.log ... X.X.X.30 is Linux, most testing done here, X.X.X.112 is Windows X.X.X.175 is Android Chrome. Some logins appear to generate more traffic on X.X.X.30, which I am assuming is a Firefox login

[JS Log.txt](https://github.com/OctoPrint/OctoPrint/files/4983399/JS.Log.txt)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/4983400/octoprint.log)

[image]

ankorite, 2020-08-01:
I can also confirm this. With Chrome web browser it seems to load then once I click anything it locks up hard. Edge seems to work ok though. I can supply logs also if interested.

foosel, 2020-08-05:
Do you happen to use KeePassXC? See #3670

brbrindle, 2020-08-05:
Yes indeed, KeePassXC is installed. 

work around is to add exception to KeePassXC in "Site Preferences"
http://"hostname"* , wild card is necessary.

Ok to close this in favor of #3670 as it more correctly discribes the problem

closed 2020-08-05 by brbrindle