OctoPrint/OctoPrint #3659: M28 Checksum error

CmdrCody51, 2020-07-28:
#### What were you doing?

1. Connect to printer
2. Goto Terminal Tab
3. In the Send box:
	M28 NEWTEST.GCO		; any file name will do (Octoprint responds with manual entry blurb)
        Detected manual streaming. Disabling temperature polling. Finish writing with M29.
        Do NOT attempt to print while manually streaming!
	Recv: echo:Now fresh file: NEWTEST.GCO
	Recv: Writing to file: NEWTEST.GCO
	Recv: ok
	G91			; any command will do

#### What did you expect to happen?

I expected the command to be written to the SD card file opened with M28.

#### What happened instead?

Error:No Checksum with line number, Last Line: 1
Recv: Resend: 2
SIDE NOTE: Temperature polling never stopped either.

#### Did the same happen when running OctoPrint in safe mode?

Yes.
SIDE NOTE: Temperature polling never stopped either.

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

Raspbian Buster on Raspberry Pi 3B+

#### Printer model & used firmware incl. version

ANET A8 V1.2 board Marlin 2.0.5.3

#### Browser and version of browser, operating system running browser

Firefox 78.0.2

#### Link to octoprint.log

https://www.dropbox.com/sh/i5eedo7ctr7kwpl/AABy11Zr0x-wvUEFSfmIIlKja?dl=0
(it is a whole subdirectory both files for normal and safe)

#### Link to contents of terminal tab or serial.log

https://www.dropbox.com/sh/i5eedo7ctr7kwpl/AABy11Zr0x-wvUEFSfmIIlKja?dl=0
(it is a whole subdirectory both files for normal and safe)

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

BeatSlayer, 2020-07-29:
Isn't the checksum missing???

CmdrCody51, 2020-07-30:
> Isn't the checksum missing???

Yes it is. That's why Marlin is rejecting the data.
I also noticed on manually entered data the checksum is never included in the serial.log so OctoPrint must be calculating it after the logging process. Except when you do a M28 - ergo a bug. When running a program, OctoPrint logs the checksum sometimes. Like after a N0 M110 N0*125. but not when it sends M155 S2? So I'm confused and so is Marlin.

foosel, 2020-08-04:
Looks like your copy of Marlin requires line numbers and checksums to be enabled if you are writing to a file. That is a new one for me, so far I only had cases of "firmware always requires checksums" or "firmware doesn't enforce checksums", not "firmware enforces checksums while writing to SD". 

Easy workaround though:

  * Go to Settings > Serial Connection > Firmware & Protocol
  * Uncheck "Enable automatic firmware detection"
  * Set "" to "Always"

That will send all commands with line numbers and checksums and thus should no longer cause you issues during streaming. I'll try to figure out if that is now general Marlin behaviour that might require a new default behaviour.

> I also noticed on manually entered data the checksum is never included in the serial.log so OctoPrint must be calculating it after the logging process.

That assumption is wrong. What is in `serial.log` is *exactly* what goes and comes over the line. If something is in there, it went over serial this way. If something isn't in there, it didn't go over serial.

foosel, 2020-08-04:
Have pushed a commit that should enforce checksums when manually streaming as well. Will be part of 1.4.2.

commit 1d55e488 (2020-08-04): 🥅 Enable checksum when manually streaming

Apparently some firmware variants (?) out there now require that, and
it actually makes sense in general.

Closes #3659
closed 2020-11-30 by foosel