OctoPrint/OctoPrint #3660: GCode script for event 'After connection to printer is established' sent before Octoprint final M876 command

bergdahl, 2020-07-28:
#### What were you doing?

I am trying to get a good workflow for the Prusa Mini, but it disables the main menu when Octoprint is connected, ie when it gets the M876 P1 command at connect time. That command disables the normal menu options such as for changing filament, etc. I tried to be smart about it and add a `M876 P0` command for the _After connection to printer is established_ setting, but it turns out it sent before Octoprint is sending it's own `M876 P1` command after this event.

There is a corresponding bug reported for the Mini's firmware, the principal problem seems to be that the firmware has no way of detecting if the serial connection is up or not, so it has to rely on the `M876` command. 
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/360

#### What did you expect to happen?

Prusa Mini should return to the normal menu system. 

#### What happened instead?

Prusa Mini is stuck in the connected to Octoprint mode.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.3.12

#### Operating System running OctoPrint

Octopi 0.17.0

#### Printer model & used firmware incl. version

Prusa Mini firmware version 4.1.0-final

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

https://gist.github.com/bergdahl/cd3d3f5b174c0fdd42ad12243447c4ed

#### Link to contents of terminal tab or serial.log

https://gist.github.com/bergdahl/b0f7db460e5c1523e3d532ab146d6445

I have read the FAQ.

foosel, 2020-07-28:
Try setting Action Command Prompt support to Never in Settings > Plugins > Action Command Prompt. Or alternatively disable that plugin altogether. It's the thing sending the command in question in completely unrelated to the GCODE script triggering. The script in question being triggered prior to anything else is working as intended and not a bug.

It's very strange for the firmware to hook their "OctoPrint detection" into such a command to begin with, and frankly I'd consider it broken behaviour to be locked out of the regular printer menu just because a serial host is connected, at least not if that can be disabled.

bergdahl, 2020-07-29:
Setting the _Action Command Prompt_ to never indeed solves the problem. I will forward your comments to the Prusa team.

foosel, 2020-07-29:
Closing then as there's a workaround and OctoPrint is behaving correctly here.

closed 2020-07-29 by foosel