OctoPrint/OctoPrint #3662: [RC Feedback] Feedback on the 1.4.1rc4 Release Candidate

foosel, 2020-07-29:
Please provide *general* feedback on your experience with the 1.4.1rc4 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

  * ❓ ["when upload to SD, the heaters turn on with the g-code specified temp"](https://github.com/OctoPrint/OctoPrint/issues/3643#issuecomment-660105814) - awaiting full bug report

### Third party issues

  * fabianonline/OctoPrint-Telegram#269 - Telegram plugin doesn't load anymore - plugin uses long deprecated imports, PR submitted
  * BillyBlaze/OctoPrint-TouchUI#413 - Misplaced buttons and a missing file list on 1.4.1rc3 - TouchUI plugin issue, PR merged
  * #3650 - Free/Used Space is Blank on rc3. and Unable to successfully upload gcode - GcodeViewer plugin issue, filed as ieatacid/OctoPrint-GcodeEditor#17
  * ["There might be a bug with the serial port detection. When I reset my board (or do a firmware upgrade), Octoprint does not connect until [I] restart [it]."](https://github.com/OctoPrint/OctoPrint/issues/3643#issuecomment-660145814) - appears to be caused by OctoScreen somehow, see https://github.com/OctoPrint/OctoPrint/issues/3643#issuecomment-662097611
  * NotExpectedYet/OctoFarm#85 - OctoPrint 1.4.1 and Websocket Error - looks like an implementation error of the socket authentication on part of OctoFarm that so far was covered by a bug in OctoPrint

### Issues needing analysis

b-morgan, 2020-07-29:
Upgraded from rc3 to rc4 without incident.

cp2004, 2020-07-30:
Was already running `staging/maintenance` to check the fix to my report #3653, now updated to 1.4.1rc4 on my Pi.
All problems reported appear solved, connection abort is a lot faster for me now!

ChrisHeerschap, 2020-07-31:
Upgrade from rc3 to rc4 no problems. Have run a number of prints on my MK3S with no complaints, haven't tested with the Prusa Mini. Will soon have a third Octoprint install for my Anycubic - if we're still on the RC at that point I'll test on that. So far I haven't seen anything that makes me think this isn't release ready. Thank you!

DodgeDeBoulet, 2020-08-02:
Upgraded with no issue, but Close button does not seem to work to exit OctoPrint settings. The "x" in the upper right hand corner works, though. Tested in both Chromium-based Edge and Firefox browsers.

foosel, 2020-08-03:
> Upgraded with no issue, but Close button does not seem to work to exit OctoPrint settings.

@DodgeDeBoulet Cannot confirm against a stock copy here. Make sure it's not a plugin issue (start in safe mode).

DodgeDeBoulet, 2020-08-03:
Restarted in Safe mode, Close button in settings works normally. Restarted again in Normal mode, Close button in settings **still** works normally.

I had restarted OctoPrint previously and the problem persisted, but did not use Safe mode. I will continue to investigate.

schnello, 2020-08-03:
Upgraded from RC3 to RC4 and perform some smaller prints. All OK and works fine.

Danke

foosel, 2020-08-03:
Stable release scheduled for tomorrow.

JohnOCFII, 2020-08-03:
All good so far:

Upgraded (from 1.4.0 to 1.4.1rc1 to 1.4.1rc2 to 1.4.1rc3to 1.4.1rc4) with no issues
Uploaded new GCODE
Printed GCODE on Prusa MK3S/wMMU2 in single extruder mode
GCODE viewer worked well
Terminal worked well
Timelapse worked well
Prusa MMU2 Select Filament plugin worked well
OctoSlack plugin worked well
Print Job History plugin worked well

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel