OctoPrint/OctoPrint #3664: [Request]: Update Discovery plugin to be python3 compatible

zeonin, 2020-07-31:
**Is your feature request related to a problem? Please describe.**

Right now the built-in Discovery plugin depends on pybonjour for Bonjour/Zeroconf, which is only compatible with python2. As python2 has been deprecated and OctoPrint has been moving towards python3 compatibility, it would be nice to not lose support for Zeroconf.

**Describe the solution you'd like**

[This line](https://github.com/OctoPrint/OctoPrint/blob/e9b6d22932a9baa81668dab884fce4b2dbe384a7/src/octoprint/plugins/discovery/__init__.py#L23) in the plugin suggests either porting pybonjour to python3 or to find an alternative. Personally, I would like to see a library that has had active contributions in the past decade as opposed to porting pybonjour. A quick search has turned up [python-zeroconf](https://github.com/jstasiak/python-zeroconf) as a potential replacement that is actively developed.

**Describe alternatives you've considered**
Alternatively, pybonjour could be modernized enough to run against python3. It [seems there was a previous attempt](https://github.com/depl0y/pybonjour-python3), though it doesn't seem that it's been touched in about 7 years.

**Additional context**
None at the moment, though I might try to give it a shot updating the plugin to python-zeroconf.

fieldOfView, 2020-07-31:
python-zeroconf is what is used in Cura, though Cura uses it for discovery as opposed to advertising itself (which is what OctoPrint would do). It is actively maintained and the maintainers are open to (well-tested) PRs if needed.

foosel, 2020-07-31:
Looks like porting over to python-zeroconf is fairly straight forward, I just threw around 2h at it and it's already looking very shippable. Just need to test the discovery helpers.

`maintenance` branch/1.4.x will have to stick to python-zeroconf 0.19.1 as that's the last version to officially support Python 2 (which 1.4.x still needs to support). On `devel`/2.0.0 I'll use a current version.

foosel, 2020-07-31:
Well, that was easy. Implemented on `maintenance` and ready for 1.4.2.

fieldOfView, 2020-07-31:
From experience with Cura, there might be some serious performance issues on certain networks with versions 0.18-0.24:
https://github.com/jstasiak/python-zeroconf/issues/133

These were fixed here:
https://github.com/jstasiak/python-zeroconf/pull/205, released as 0.24.1

Would you be open to a solution where Python 2 instances use pybonjour and Python 3 instances use python-zeroconf?

foosel, 2020-07-31:
It looks like this only relates to browsing, which is only exposed via a helper and not used by any bundled plugins. So the risk of this causing issues is probably fairly small.

An alternative to a python version specific switch would be to vendor bundle it for 1.4.x and apply the changes from the PR to the bundled version.

fieldOfView, 2020-07-31:
> So the risk of this causing issues is probably fairly small.

👍 
Even so it is probably a good idea to ask for feedback specifically from users with many (hundreds) Apple devices on their network once 1.4.2 RC comes around. Unless you feel you need a reason to buy 100 iphones to set up testing this yourself ofcourse 😉.

foosel, 2020-07-31:
So, slightly changed [https://github.com/OctoPrint/OctoPrint/compare/cae900b1fb08...ffde393c666a](https://github.com/OctoPrint/OctoPrint/compare/cae900b1fb08...ffde393c666a). Vendored lib for Py2, adjusted by the aforementioned performance patch plus some minor equality check fixes and also a bit of backporting of some method signatures to allow one client code for both the vendored lib and the officially maintained one. The latter will be automatically used under Py3.

foosel, 2020-08-06:
Make that 1.4.3, 1.4.2 just got turned into a hotfix release.

foosel, 2020-08-10:
Actually make that 1.5.0 because I've decided to finally *fully* adhere to semver and maintenance releases will see a minor increase in the future instead of patch, patch will be used for hotfixes if needed.

commit 06637280 (2020-07-31): ♻ Discovery: Switch to python-zeroconf

Pybonjour is no longer maintained and Python 2 only.

Closes #3664
closed 2020-11-30 by foosel