OctoPrint/OctoPrint #3666: ImportError: No module named pigpio

DjSteker, 2020-07-31:
**I found this:**

```
 File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_GSLC/__init__.py", line 4, in <module>
    import pigpio
ImportError: No module named pigpio

```

**Is it as easy as adding the GPIO library in the folder?**

foosel, 2020-07-31:
That's a problem with a third party plugin you installed, not OctoPrint. Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin **repository**, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
>
>   * missing or broken homepage, source or download URLs
>   * broken images
>   * incorrect compatibility information
>   * ... in short, anything related to the data on plugins.octoprint.org
>
> What you should not report here:
>
>   * Any kinds of issues with the plugins themselves - report those
>     to the plugin author instead!

Thanks :)

closed 2020-07-31 by foosel

fieldOfView, 2020-07-31:
The tracker for the plugin in question can be found here: https://github.com/Skiepp/GCodeSuperLaserController

DjSteker, 2020-07-31:
okay
thank you :)
Do I delete the message? because the problem is several plugins

fieldOfView, 2020-07-31:
The issue is "closed", and that suffices.

If the problem occurs in several plugins, it will have to be fixed in each plugin individually. If one of those plugins comes bundled with OctoPrint (not OctoPi!), please let us know here.

DjSteker, 2020-08-01:
This is what comes out of my console (the program works well and I can use it and even plugins except the photo plugin per layer).     _and just that plugin wantedmake a modification "that takes the photo without interrupting anything"_

```
pi@raspberrypi:~ $ ~/OctoPrint/venv/bin/octoprint serve
2020-07-31 02:16:35,110 - octoprint.startup - INFO - ******************************************************************************
2020-07-31 02:16:35,113 - octoprint.startup - INFO - Starting OctoPrint 1.4.0
2020-07-31 02:16:35,113 - octoprint.startup - INFO - ******************************************************************************
2020-07-31 02:16:35,911 - octoprint.startup - INFO - We don't appear to be online, not fetching plugin blacklist
2020-07-31 02:16:35,996 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-07-31 02:16:37,843 - octoprint.plugin.core - ERROR - Error loading plugin GSLC
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 967, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_GSLC/__init__.py", line 4, in <module>
    import pigpio
ImportError: No module named pigpio
2020-07-31 02:16:38,069 - octoprint.plugins.octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2020-07-31 02:16:38,095 - octoprint.plugin.core - INFO - Found 24 plugin(s) providing 24 mixin implementations, 41 hook handlers
2020-07-31 02:16:38,237 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-07-31 02:16:38,378 - octoprint.server - INFO - Intermediary server started
2020-07-31 02:16:38,380 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-07-31 02:16:38,541 - octoprint.plugin.core - ERROR - Error loading plugin GSLC
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 967, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_GSLC/__init__.py", line 4, in <module>
    import pigpio
ImportError: No module named pigpio
2020-07-31 02:16:38,548 - octoprint.plugin.core - INFO - Found 24 plugin(s) providing 24 mixin implementations, 41 hook handlers
2020-07-31 02:16:38,584 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2020-07-31 02:16:39,018 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2020-07-31 02:16:39,024 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: u"Need(method='role', value=u'plugin_action_command_prompt_interact')")
2020-07-31 02:16:39,025 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: u"Need(method='role', value=u'plugin_announcements_read')")
2020-07-31 02:16:39,026 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: u"Need(method='role', value=u'plugin_announcements_manage'), Need(method='role', value=u'plugin_announcements_read')")
2020-07-31 02:16:39,027 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: u"Need(method='role', value=u'plugin_appkeys_admin')")
2020-07-31 02:16:39,029 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: u"Need(method='role', value=u'plugin_backup_access')")
2020-07-31 02:16:39,030 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: u"Need(method='role', value=u'plugin_logging_manage')")
2020-07-31 02:16:39,031 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: u"Need(method='role', value=u'plugin_pi_support_check')")
2020-07-31 02:16:39,032 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: u"Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-07-31 02:16:39,034 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: u"Need(method='role', value=u'plugin_pluginmanager_install'), Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-07-31 02:16:39,036 - octoprint.server - INFO - Added new permission from plugin printer_safety_check: PLUGIN_PRINTER_SAFETY_CHECK_DISPLAY (needs: u"Need(method='role', value=u'plugin_printer_safety_check_display')")
[... cut ...]
2020-07-31 02:16:44,431 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.0.100:5000/'}
2020-07-31 02:16:54,328 - octoprint.server.preemptive_cache - INFO - ... done in 9.90s
^C2020-07-31 02:17:11,948 - octoprint.server - INFO - Shutting down...
2020-07-31 02:17:12,583 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-07-31 02:17:12,587 - octoprint.events - INFO - Event loop shut down
2020-07-31 02:17:12,628 - octoprint.server - INFO - Goodbye!

```

foosel, 2020-08-01:
As already said, that is a problem with a plugin, not OctoPrint. You need to take this up with the plugin author.