OctoPrint/OctoPrint #3668: Plugin install appears to timeout but continues in background

KenLucke, 2020-08-01:
#### What were you doing?

Installing OctoPrint-Stats  plugin
I'm not seeing any activity on this plugin in its own repository, but I left the same information there...  It appears more of a plugin manager bug than a plugin bug anyway.

I've been trying for some time to install this plugin on OctoPrint 1.4.0 running on a Raspberry Pi 3b+,  but it always failed with the following in the install dialog (long time between each of the "still running" messages (note:  it installed just fine, albeit slowly with the "still running" notations, on 2 different RPi 4B units running the same OctoPrint version):

```
.
.
.
Building wheels for collected packages: OctoPrint-Stats, pandas
Building wheel for OctoPrint-Stats (setup.py): started
Building wheel for OctoPrint-Stats (setup.py): finished with status 'done'
Created wheel for OctoPrint-Stats: filename=OctoPrint_Stats-2.0.2-py2-none-any.whl size=870537 
sha256=1665f6c5bdadfa85320065f12537ead3eb33f2f010be1ab5205deb005c1449f2
Stored in directory: /tmp/pip-ephem-wheel-cache-6TqxHu/wheels/59/0c/e9/34b9eae468685a3935f78c4f864d5f9ebf3e6fdbc139515cf2
Building wheel for pandas (setup.py): started
Building wheel for pandas (setup.py): still running...
Building wheel for pandas (setup.py): still running...
Building wheel for pandas (setup.py): still running...
Building wheel for pandas (setup.py): still running...
Error!Could not install plugin, unknown error, please consult octoprint.log for details

```
It also threw the "There was a problem" dialog up upper right screen.

In the past, when this had happened, it had hung OctoPrint so hard that I had to do an SSH into the server and do a full reboot, not just a service restart, and when things came up, there was nothing in the logs - literally nothing other than the restart logging (because I had always deleted the logs so I could see what was going on after the first fail).  I am ass/u/ming that that's because when it hung, it didn't get a chance to flush the data to the file.

Yesterday, I did my monthly "apt-get update"s and "apt-get upgrade"s to all my RPis, so I thought I'd give it it a try again.  I figured if it failed this time, I'd swap the SD card to one of the Pi 4Bs and install it there, then swap back.  Lo and behold, surprise, surprise, surprise, it failed once again, timing out during the last part of the install.... or so it seemed:  I got the same message in the install progress box and the warning dialog upper right, but things were not locked up this time, so I was able to exit out of the dialog boxes and back to the main screen.  As it was late at night, and I didn't want to traipse downstairs to do the card swap I had planned, I just left things as they were.

Later in the night, I had occasion to look at the computer, and, wonder of wonders, it showed a dialog upper right saying "Successfully installed, restart required"!   I restarted OctoPrint and there it was, installed and running.

Somehow, it's appearing as if it is timing out in the install, but CONTINUING to install in the background.

#### What did you expect to happen?
Should have installed

#### What happened instead?
Install appears to have timed out, but actually continued in the background

#### Did the same happen when running OctoPrint in safe mode?
Did not try

#### Version of OctoPrint
OctoPrint version : 1.4.0
OctoPi version    : 0.17.0

#### Operating System running OctoPrint
Linux / Raspberry Pi 3b+
Linux 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l

#### Printer model & used firmware incl. version
n/a

#### Browser and version of browser, operating system running browser
n/a

#### Link to octoprint.log
plugin manager log fragment below (note that there are 26 "still running"s in this log, the copy/paste above from the status display only had 4 before the timeout):

```.
.
.
2020-08-01 01:31:23,158 > Building wheels for collected packages: OctoPrint-Stats, pandas
2020-08-01 01:31:23,159 > Building wheel for OctoPrint-Stats (setup.py): started
2020-08-01 01:31:25,164 > Building wheel for OctoPrint-Stats (setup.py): finished with status 'done'
2020-08-01 01:31:25,165 > Created wheel for OctoPrint-Stats: filename=OctoPrint_Stats-2.0.2-py2-none-any.whl size=870537 sha256=1665f6c5bdadfa85320065f12537ead3eb33f2f010be1ab5205deb005c1449f2
2020-08-01 01:31:25,165 > Stored in directory: /tmp/pip-ephem-wheel-cache-6TqxHu/wheels/59/0c/e9/34b9eae468685a3935f78c4f864d5f9ebf3e6fdbc139515cf2
2020-08-01 01:31:25,166 > Building wheel for pandas (setup.py): started
2020-08-01 01:32:25,532 > Building wheel for pandas (setup.py): still running...
2020-08-01 01:33:34,725 > Building wheel for pandas (setup.py): still running...
2020-08-01 01:34:34,784 > Building wheel for pandas (setup.py): still running...
2020-08-01 01:41:28,172 > Building wheel for pandas (setup.py): still running...
2020-08-01 01:54:45,815 > Building wheel for pandas (setup.py): still running...
2020-08-01 01:56:08,896 > Building wheel for pandas (setup.py): still running...
2020-08-01 01:58:11,424 > Building wheel for pandas (setup.py): still running...
2020-08-01 01:59:52,186 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:01:51,304 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:03:10,391 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:10:38,823 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:12:04,933 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:13:14,380 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:16:48,572 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:18:02,183 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:19:05,501 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:20:09,770 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:21:15,821 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:22:55,162 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:24:07,219 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:25:24,597 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:26:39,663 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:28:48,192 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:31:18,849 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:32:29,986 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:34:56,472 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:36:05,554 > Building wheel for pandas (setup.py): still running...
2020-08-01 02:36:46,075 > Building wheel for pandas (setup.py): finished with status 'done'
2020-08-01 02:36:46,076 > Created wheel for pandas: filename=pandas-0.24.2-cp27-cp27mu-linux_armv7l.whl size=27967158 sha256=840b9f64a1e428c74079cd13c1712dc92e5c2f84936df68070cf7364f99c74bf
2020-08-01 02:36:46,077 > Stored in directory: /tmp/pip-ephem-wheel-cache-6TqxHu/wheels/5f/b4/0b/b1670f56250abdd8f7f200c1a63fbb96e2132f2ed7ee1f8b71
2020-08-01 02:36:46,077 > Successfully built OctoPrint-Stats pandas
2020-08-01 02:36:47,194 > Installing collected packages: tinydb, python-dateutil, pandas, OctoPrint-Stats
2020-08-01 02:38:04,739 > Successfully installed OctoPrint-Stats-2.0.2 pandas-0.24.2 python-dateutil-2.8.1 tinydb-3.15.2
```

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-09-03:
Thanks for reporting this. 

The issue is that the plugin installation request in the plugin manager times out. For some reason it's still assuming a synchronous install that finishes during the request run time. That's a bug, it needs to be changed so it triggers the install asynchronously and then returns (possibly with a status URL). I'll have to look into this.

foosel, 2020-09-07:
Fixed pushed to `maintenance`, will be included in 1.5.0.

commit 4cf2a673 (2020-09-07): 🐛 pmgr: turn install API command fully async

Fixes #3668
closed 2020-11-30 by foosel