OctoPrint/OctoPrint #3669: [Request] Make top bar fixed, not scrolling

Jeff5263, 2020-08-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I WOULD LIKE TO SEE THE TOP BAR (WITH MY PROFILE, SETTINGS, SHUTDOWN, ETC.) FIXED SO IT DOESN'T SCROLL WITH THE REST OF THE PAGE. -THANKS

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jneilliii, 2020-08-01:
Install the [Floating Navbar](https://plugins.octoprint.org/plugins/floatingnavbar/) plugin.

foosel, 2020-08-03:
Like @jneilliii said, there's a plugin for that. Also, please don't use all caps, it's reads as if you were screaming and hence gives a quite rude first impression ;)

closed 2020-08-03 by foosel