OctoPrint/OctoPrint #3670: KeePassXC Chrome plugin prevents OctoPrint page from loading

sschueller, 2020-08-03:
#### What were you doing?
When using the chrome KeePassXC plugin the OctoPrint UI no longer loads. It doesn't matter if the plugin is correctly configured or not. The Issue may be with the plugin and not OctoPrint but I wanted to make you aware of it as it wasn't easy to track down. So far I have only had issues with loading the OctoPrint UI, no other webpage has given any issues. I will open a bug report at KeePassXC as well.

1. Open Chrome, install https://chrome.google.com/webstore/detail/keepassxc-browser/oboonakemofpalcgghocfoadofidjkkk (no need to configure)
2. Browse to your OctoPrint Instance
3. Page gets stuck at loading UI
4. If In safemode you will get to the page but it will freeze after the first action

#### What did you expect to happen?
OctoPrint webpage loads

#### What happened instead?
Page freezes, browser complains about hanging page.

#### Did the same happen when running OctoPrint in safe mode?
Page loads but will freeze after first attempt to click an item.

#### Version of OctoPrint
OctoPrint 1.4.0

#### Operating System running OctoPrint
OctoPi 0.15.1

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
OSX: 10.15.5 (19F101)
Chrome Version 84.0.4147.89 (Official Build) (64-bit)
Also happends on:
Windows 10
Chrome Version 84.0.4147.105 (Official Build) (64-bit)

#### Link to octoprint.log
https://gist.github.com/sschueller/8757ecb132e65f7a25bb80660f22d41b

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
```
Starting dependency resolution...
packed_core.js?3514a079:17074 ... dependency resolution done
packed_core.js?3514a079:17384 Initial application setup done, connecting to server...
packed_core.js?3514a079:15010 Connected to the server
packed_core.js?3514a079:17349 Finalizing application startup
packed_core.js?3514a079:17224 Going to bind 42 view models...
packed_core.js?3514a079:17277 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?3514a079:17277 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?3514a079:17277 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?3514a079:17277 Did not bind view model bedlevelvisualizerViewModel to target #wizard_plugin_bedlevelvisualizer since it does not exist
packed_core.js?3514a079:3334 User sschueller logged in
packed_core.js?3514a079:17316 ... binding done
packed_core.js?3514a079:17328 Application startup complete
``` 

#### Screenshot(s)/video(s) showing the problem:
[image]

- [x] I have read the FAQ.

foosel, 2020-08-03:
Your KeepassXC ticket was a duplicate of keepassxreboot/keepassxc-browser#944. As far as I understand that ticket, the issue seems to lie with the way that KeepassXC detects fillable input fields vs OctoPrint's HTML or possibly DOM. 

I'm not sure there's anything I can do from my side about this that wouldn't mean a complete rewrite of the UI or at the very least quite disruptive changes, and considering that a workaround for the problem is provided in the aforementioned ticket, I'll be marking this as "not octoprint" for now.

It's interesting that it behaves slightly less drastically in safe mode, meaning that whatever happens is made worse by one of the third party plugins you have installed. If you can figure out which one, it might help to narrow down on the core cause.

sschueller, 2020-08-05:
The multicam plugin (https://github.com/mikedmor/OctoPrint_MultiCam) appears to be what makes the difference. Disabling it gets me as far as in safemode.

brbrindle, 2020-08-05:
With no additional other than standard plugin's, The page loads but anything that requires a pop up crashes the page.

foosel, 2020-08-05:
I've chimed in on the KeePassXC Chrome plugin to hopefully get an idea of what that plugin is even doing that might cause these issues, but that's about all I can do for now.

@brbrindle does the workaround you posted in https://github.com/OctoPrint/OctoPrint/issues/3657#issuecomment-669246355 work in general or do popups still make it crash? I'm unclear on this here. Also, what do you qualify as popups? Modals (e.g. the settings dialog), notifications (e.g. update notifications), something else entirely?

brbrindle, 2020-08-05:
yes work around I posted seems to work. 
Yes you are correct, settings notifications and confirmation buttons (cancel) all seem to either not load entirely (notifications, setup wizard) or crash the webpage (settings and confirmation buttons).

foosel, 2020-08-05:
So anything that overlays the page? Weird... Let's hope we'll get some pointers from the KeePassXC plugin maintainers.

foosel, 2020-09-03:
It looks like the ticket got closed on KeePassXC's side. Apparently one can now disable "dynamic input field detection" and that has solved issues for at least one user in another piece of affected software.

Could anyone able to reproduce this on OctoPrint please check if this is still an issue with current versions of KeePassXC and if it is report that back **in the KeePassXC ticket** as it's currently considered solved over there?

sschueller, 2020-09-03:
I just tested it in Chrome and Brave. It works now even with "dynamic input field detection" enabled. 

```
KeePassXC-Browser Version: 1.7.0
KeePassXC Version: 2.6.1
Brave: Version 1.11.104 Chromium: 84.0.4147.105 (Official Build) (64-bit)
OctoPrint 1.4.2 
Python 2.7.13 
OctoPi 0.15.1
```

Thanks 👍

foosel, 2020-09-03:
Don't thank me, thank the KeePassXC devs ;)

Closing this then.

closed 2020-09-03 by foosel