OctoPrint/OctoPrint #3673: Set default webcam settings button [Request]

mattbruman, 2020-08-04:
Please add a reset default webcam settings button, I have on many occasion tried different webcam setups and actually this time it was a bug with multicam and my default was overwritten with one of the multicam cams. Would be nice to be able to click reset to default without having to remember or search for the default string. This is a nice to have request not really process breaking.

foosel, 2020-08-04:
There are no default webcam settings for OctoPrint in general that such a button could restore 😕 

There are only default webcam settings for the bundled mjpg streamer on OctoPi, and implementing something that restores those in core would be wrong considering that not everyone runs OctoPi. That's also why the FAQ talks about the default settings *on OctoPi*.

I honestly can't think of a way to do this cleanly without pulling in OctoPi specifics into core OctoPrint that simply don't belong there or writing complex plugin hooks to be able to override default settings from platform specific plugins (to be combined with a dedicated OctoPi config plugin).

mattbruman, 2020-08-04:
What about storing the previous entered strings and then just have "store as default" button? Was just a minor thing no big deal if its too much. ¯\_(ツ)_/¯