OctoPrint/OctoPrint #3674: octoprint 1.4.1 /plugin/pluginmanager/upload_file 400 (Bad Request)

devildant, 2020-08-04:
#### What were you doing?

Install new plugin from an uploaded file return an error

inside the console i can see this error : 
[image]

[image]

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-08-04:
Logs please?

devildant, 2020-08-04:
Hello @foosel, i don't have any log, during the upload i open run tail command on octoprint.log, and nothing :(

devildant, 2020-08-04:
I try disable all my additional plugin => failed
i force re-install octo 1.4.1 => failed
i clear my cache => failed
i re-install octo 1.4.0 => work well

devildant, 2020-08-04:
example video :
[20200804_2131_19.zip](https://github.com/OctoPrint/OctoPrint/files/5024166/20200804_2131_19.zip)

jneilliii, 2020-08-04:
There could be additional information in the log that identifies the root cause, not just while you are trying to do it. I suspect she might want both the octoprint.log and plugin_pluginmanager_console.log.

jneilliii, 2020-08-04:
and possibly the zip file you're trying to upload as well.

devildant, 2020-08-04:
I doubt it really helps, well it costs nothing.
I deactivated my plugins while dev (because bcp of useless log)
and I redid the test
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5024290/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/OctoPrint/OctoPrint/files/5024291/plugin_pluginmanager_console.log)

devildant, 2020-08-04:
New test : 
i install an octopi on new SD
1) configure octoprint => OK
2) try to install plugin from file => OK
3) i update octoprint to 1.4.1 => OK
4) try again to install same plugin from file => KO

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5024472/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/OctoPrint/OctoPrint/files/5024474/plugin_pluginmanager_console.log)
[plugin_softwareupdate_console.log](https://github.com/OctoPrint/OctoPrint/files/5024475/plugin_softwareupdate_console.log)

on the plugin_pluginmanager_console.log, i see only one install, the second in 1.4.1 is missing

jneilliii, 2020-08-04:
What about the plugin you're trying to install from zip? Is that available for additional testing outside of your setup?

devildant, 2020-08-04:
@jneilliii mines, yours, touchui plugins....

jneilliii, 2020-08-04:
touchui just released an update a little while ago and I know it sounded like it had some nasty stuff in it prior to update. Force an upgrade check and make sure you have 0.3.15 installed.

devildant, 2020-08-04:
my touchui is up to date I pulled it a few hours ago from the maintenance branch.
but that's not the problem ... i use the zip touch ui as it is the closest in my file list. No matter the zip, the upload returns a 400 systematically.
which bothers me a lot for the development of my plugins because I tests before commit/push.
noted that there is no problem when I downgrad octoprint to 1.4.0
I think @foosel  will be the most apt to understand the anomaly.
I ruled out the problem due to my configuration, with a fresh install (comment above)
I would be available if we need to test :)

jneilliii, 2020-08-05:
Agreed foosel will probably be the only one to possibly figure out what's going on here. I just tested on two separate pis, one with Python 3.7.3 and one with Python 2.7.16 both running 1.4.1 and was able to install a plugin I had downloaded from github that I had never installed before without issue.

devildant, 2020-08-05:
@jneilliii it's really strange because even with a new octopi install the problem persists on my side

foosel, 2020-08-05:
I can reproduce it on both Firefox and Chrome. The file upload button seems to generate an invalid file upload request that gets nuked at tornado level, before entering regular upload processing, which might explain the lack of logs. I have not yet figured out why though, and I also can't remember modifying this in any way other than also allowing .py files (for single file plugin install). It's a bit strange at the moment.

In any case, considering that this is probably a rarely used feature outside of development (I guess it is somewhat telling that no one reported this during the month long RC phase) and there exist alternatives (upload through URL, install via command line), I will not be pushing out a hotfix for this.

devildant, 2020-08-05:
@foosel 
I unfortunately did not have time to participate in the RC this time :(
now, I have to find the command line to install a plugin with pip ^^

devildant, 2020-08-05:
maybe ~/oprint/bin/pip plugin.zip ^^

foosel, 2020-08-05:
Alternatively you could also fire up a web server serving whereever you store your plugin zip during development and install from that. `python -m SimpleHTTPServer 8000` (Python 2) or `python -m http.server 8000` (Python 3) can be very helpful here.

devildant, 2020-08-05:
@foosel yes good idea :)

foosel, 2020-08-05:
Ok, issue identified. The upload endpoint was renamed from `upload_archive` to `upload_file` (as you no longer can upload only plugin archives but also single file plugins). The problem: I forgot to also modify the upload endpoint in the bodysize hook. So now the upload is denied due to a too large file size. Easy to fix in the code, and also easily fixed by increasing the general max upload size via `server.maxSize` in `config.yaml`:

```yaml
server:
  maxSize: 52428800
```

After that it should work all as before.

foosel, 2020-08-05:
Fix pushed to `maintenance` and ready for 1.4.2

devildant, 2020-08-05:
thx @foosel,
i added the maxSize, it's work perfectly.
during the RC of 1.4.2 I would try to participate this time: p (I attract bugs like a magnet: p)

foosel, 2020-08-05:
Remember to share logs then though :P

devildant, 2020-08-05:
yes I would, but with each anomaly, I check the logs to see if they are relevant, and if I can give you the info more quickly.
here not seeing any startup error and not having seen a suspicious log, with the fact that when I encountered the error no log was not generated, I did not estimate their usefulness, probably wrong.
but I understand :)

foosel, 2020-08-05:
I always need logs since they don't just tell me something about a direct error, but also about your environment, what other, on first glance unrelated, things happened before and so on. I keep asking for them so insistently for a reason. You are welcome to check them yourselves and make guesses about what might be the relevant lines, but *always* also share the full log because the whole log is as a whole important data about the environment the bug happens in, even if it doesn't show up in the log itself.

devildant, 2020-08-05:
I understand everything, it's also a deformation related to my work (I dev in java and the servers are hosted in my company, so I know the environment, and the logs are so large that we target the data of the log.
octoprint being hosted by the client, it's different because it is important to know the context)
I'm going off topic :)
I would be careful to pass them on

foosel, 2020-08-06:
Good news, I'll push the fix for this out with a hotfix release I'll do for #3678 and #3676

commit 5a33ecf6 (2020-08-05): 🐛 pmgr: fix max upload size

Closes #3674
commit 8c829048 (2020-08-06): 🐛 pmgr: fix max upload size

Closes #3674
closed 2020-08-06 by foosel