OctoPrint/OctoPrint #3675: File manager bug in release 111.4.1

paul30003, 2020-08-04:
#### What were you doing?

Chrome used on windows 10

1. Drag a file to upload.
2. Saving progress bar remains and saved file is not displayed in list
3. Refresh and the file then appears.
4. Deleting a file has similar results.
5. Pressing the delete icon does not remove the file from the list.
6. Refresh and the file disappears.

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

mouse100, 2020-08-04:
I have experienced the same issue after the update. How can i solve It?

GhostlyCrowd, 2020-08-05:
Toss my hat into the ring for this issue as well, except refreshing does not show the file only rebooting the octoprint service makes the file show.

foosel, 2020-08-05:
Sadly all your reports lack some of the information that is needed in order to look into this issue as you didn't fill out the provided ticket template:

  * whether you have been able to reproduce this issue in [safe mode too](https://docs.octoprint.org/en/master/features/safemode.html). If it does not reproduce in safe mode it is not an issue with OctoPrint but one of your installed third party plugins.
  * your exact versions of OctoPrint  (https://discourse.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204/1)
  * the operating system running OctoPrint (name *and* version please)
  * the browser with which you are seeing this (underlying OS and full version please)
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)
  * screenshots or maybe a video showing the problem

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Please provide that so this issue can be analyzed and ideally reproduced and fixed. Thank you!

jneilliii, 2020-08-05:
I think I have isolated this issue to the Gcode Editor plugin. After seeing the log from the forum post above I installed each of the plugins until I was able to reproduce the error.  Error in developer console...

```
Uncaught TypeError: Cannot read property 'indexOf' of undefined
    at HTMLDivElement.<anonymous> (packed_plugins.js?b44129dd:147)
    at Function.each (packed_libs.js?df208d52:1)
    at w.fn.init.each (packed_libs.js?df208d52:1)
    at addEditButtonsToGcode (packed_plugins.js?b44129dd:131)
    at ItemListHelper.self.files.listHelper._updateItems (packed_plugins.js?b44129dd:346)
    at ItemListHelper.self.updateItems (packed_core.js?357b1a88:621)
    at FilesViewModel.self.changeFolderByPath (packed_core.js?357b1a88:106)
    at FilesViewModel.self.fromResponse (packed_core.js?357b1a88:101)
    at Object.<anonymous> (packed_core.js?357b1a88:97)
    at u (packed_libs.js?df208d52:1)
    at Object.fireWith [as resolveWith] (packed_libs.js?df208d52:1)
    at k (packed_libs.js?df208d52:1)
    at XMLHttpRequest.<anonymous> (packed_libs.js?df208d52:1)
    at XMLHttpRequest.sentryWrapped (packed_core.js?357b1a88:430)
```
Can everyone in this thread please verify if you have that plugin installed or not?

jr-air, 2020-08-05:
I had the same issue and disabling the GcodeEditor is the current workaround. 

I also finally converted to python3, so mentioning in case it matters.

foosel, 2020-08-05:
I've blacklisted the plugin on the plugin repository now. Restarting your instances should automatically disable it. Closing as this is not an OctoPrint issue but a third party plugin breaking stuff.

closed 2020-08-05 by foosel