OctoPrint/OctoPrint #3676: OctoPrint fails to start after 1.4.1 upgrade

nirvdrum, 2020-08-05:
## What were you doing?

I upgraded from 1.4.0 (installed via OctoPi) to 1.4.1 using the web prompt and after doing so, OctoPrint fails to start with the [same message](https://github.com/OctoPrint/OctoPrint/issues/3643#issuecomment-661323001) reported in the RC builds a couple weeks back. It looks like the original commenter may have dismissed it as a fluke, but it appears it wasn't an isolated incident. I've provided all the [information requested](https://github.com/OctoPrint/OctoPrint/issues/3643#issuecomment-661888394) of the original filer. Apologies in advance if this is a duplicate bug submission, but I didn't see an issue filed from the previous discussion.

Please note I had not been testing the RC builds. I encountered this issue with an upgrade from 1.4.0 installed via OctoPi.

<details>
<summary>~/.octoprint/logs/octoprint.log</summary>
<p>

```
pi@octoprint:~ $ cat ~/.octoprint/logs/octoprint.log
2020-08-04 00:21:23,030 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 00:21:23,035 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2020-08-04 00:21:23,036 - octoprint.server - INFO - OctoPrint 1.4.0
2020-08-04 00:21:23,041 - octoprint.plugin.core - INFO - 19 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  EEPROM Editor for MPSM (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_MPSelectMini
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/loginui
|  Malyan/Monoprice Connection Fix (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_malyan_connection_fix
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  PortLister (0.1.10) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_portlister
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Prusa Leveling Guide (1.0.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrusaLevelingGuide
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2020-08-04 00:21:23,063 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|      cores: 4
|      freq: 1200.0
|      ram: 917708800
|  os:
|      id: linux
|      platform: linux2
|  plugins:
|      pi_support:
|          model: Raspberry Pi 3 Model B Rev 1.2
|          octopi_version: 0.17.0
|  python:
|      pip: 19.3.1
|      version: 2.7.16
|      virtualenv: /home/pi/oprint
2020-08-04 00:21:23,063 - octoprint.server - INFO - ------------------------------------------------------------------------------
2020-08-04 00:36:23,032 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 00:51:23,034 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 01:06:23,035 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 01:21:23,039 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 01:24:36,978 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.13.30
2020-08-04 01:36:23,041 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 01:51:23,043 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 02:06:23,045 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 02:21:23,048 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 02:36:23,051 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 02:51:23,053 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 03:06:23,055 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 03:21:23,058 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 03:36:23,060 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 03:51:23,062 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 04:06:23,064 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-04 04:21:23,066 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... 845 lines cut ...]

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.4.1.

#### Operating System running OctoPrint

OctopPi 0.17.0.

```
pi@octoprint:~ $ cat /etc/octopi_version
0.17.0

pi@octoprint:~ $ cat /etc/octopi_buildbase
2019-09-26-raspbian-buster-lite.img

pi@octoprint:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

jneilliii, 2020-08-05:
Thanks for a very detailed and nearly complete issue ticket @nirvdrum. One important piece of information is missing `Did the same happen when running OctoPrint in safe mode?` as that will help determine if this is a plugin related issue or not. Also the issue bot is not happy because you removed the line `I have read the FAQ.` at the end. Add that back in and I think the bot will be happy again.

nirvdrum, 2020-08-05:
I'm sorry. It looks like I messed up following the issue template. I didn't explicitly run OctoPrint in safemode. OctoPrint seemed to do that on its own when I subsequently rebooted the machine. It's shown in the octoprint.log:

```
******************************************************************************
2020-08-04 23:42:44,146 - octoprint.startup - INFO - Starting OctoPrint 1.4.1
2020-08-04 23:42:44,148 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2020-08-04 23:42:44,150 - octoprint.startup - INFO - Reason for safe mode: problem during last startup
2020-08-04 23:42:44,153 - octoprint.startup - INFO - ******************************************************************************
```

foosel, 2020-08-05:
I still have not seen that myself (and I updated a LOT of instances over the course of the past month). So whatever this is, it must be something environmental. I just compared your `pip freeze` output with one from a freshly updated OctoPi 0.17.0 instance and apart from plugins they are completely identical.

Is your watched folder empty? If not, does it contain a file with non-ASCII characters by any chance? I just looked into the source of the dependency, and I could see an issue arise there under Python 2 with non-ASCII file or folder names.

`ls -la ~/.octoprint/watched` should tell us something more here.

foosel, 2020-08-05:
Good news, I have a reproduction. Setting the watched folder to being actively polled combined with a utf8 filename in the watched folder on startup under Python 2.7 triggers a bug in the new `watchdog` version that OctoPrint now depends on. 

As a workaround I suggest to clean up your watched folder and avoid non ASCII characters in filenames you put there for now. I'll add some more error resilience during startup to 1.4.2 and will also see if I can't somehow patch `watchdog` to not run into this in the first place. As it's a Python 2 only issue, I'm not sure if I should even bother trying to raise this upstream 🤔

On OctoPi this should take care of things:

    rm -rf ~/.octoprint/watched/*
    sudo service octoprint restart

foosel, 2020-08-05:
Workaround ready for 1.4.2. Until then, see my comment above on how to work around this issue in `watchdog` in 1.4.1. Either disable polling, or avoid non-ASCII characters.

nirvdrum, 2020-08-05:
It's been a while since I've actively done Python development, but I wrote a quick Ruby script to go through my watched folder recursively and find any filenames with non-ASCII characters. Every filename checked restricts characters to 7-bit ASCII codepoints:

```ruby
#!/usr/bin/env ruby

Dir.glob('/home/pi/.octoprint/watched/**/*') do |file|
  unless file.ascii_only?
    puts file
  end
end
```

I'll try disabling polling.

foosel, 2020-08-05:
Might also be an issue just with polling... It's a bit tricky to say, but I'm pretty sure it's something about the polling observer reading `unicode` instead of `str` via `os.listdir`, because that will make a check fail that will cause the `watchdog` code to attempt to access `name` on a string, which leads to the observed error.

nirvdrum, 2020-08-05:
I think I narrowed this down, but I can't see how this wouldn't be problematic for people with anything in "watched". settings.py [sets the file encoding to UTF-8](https://github.com/OctoPrint/OctoPrint/blob/3ab84ed7e4c3aaaf71fe0f184b465f25d689f929/src/octoprint/settings.py#L1), so all string literals in that file are of type `unicode` instead of `str`. This includes `_APPNAME` and some ad hoc usages of _~_ and _._ in various `os` calls.

I'll need to dig into the stdlib a bit more to confirm, but it looks like the `os` module adjusts to the source argument type. If given a `unicode` object, it will always return `unicode` objects. The watchdog library is being passed the watched folder name, which is of type `unicode`, so the output of `listdir` is a list of `unicode` objects, which naturally all fail the `if instanceof(entry, str)` check.

My local fix is this:

```diff
pi@octoprint:~ $ diff -c /home/pi/oprint/local/lib/python2.7/site-packages/watchdog/utils/dirsnapshot.py dirsnapshot.py 
*** /home/pi/oprint/local/lib/python2.7/site-packages/watchdog/utils/dirsnapshot.py	2020-08-05 12:32:04.641823018 -0400
--- dirsnapshot.py	2020-08-05 12:31:47.321943695 -0400
***************
*** 262,268 ****
      def walk(self, root):
          try:
              paths = [os.path.join(root, entry if isinstance(entry, str) else entry.name)
!                      for entry in self.listdir(root)]
          except OSError as e:
              # Directory may have been deleted between finding it in the directory
              # list of its parent and trying to delete its contents. If this
--- 262,268 ----
      def walk(self, root):
          try:
              paths = [os.path.join(root, entry if isinstance(entry, str) else entry.name)
!                      for entry in self.listdir(str(root))]
          except OSError as e:
              # Directory may have been deleted between finding it in the directory
              # list of its parent and trying to delete its contents. If this
```

Obviously we wouldn't want to modify the watchdog library like this. Removing the _coding_ magic comment isn't sufficient either. For one, the author name value contains non-ASCII characters and every file would need to be updated to handle that. But, even doing that locally I was finding string literals coming out as `unicode` and it's not clear why; `sys.getdefaultencoding` returns _ascii_ as expected. Using the `b` prefix to indicate byte strings looks like it'd work in the default case, but wouldn't be appropriate if the user specified a watched directory that truly contained non-ASCII characters.

Given all that, it looks to me like this is a problem with the default base directory and not having to do anything with the contents of the watched directory. Moreover, it looks to me like _any_ file at all in the watched directory would be enough to trigger this problem. But, there wasn't widespread reports of the problem so I'm a bit skeptical of that.

foosel, 2020-08-06:
I was pretty sure I had tested with a ascii-only file in the watched folder with polling enabled and that didn't fail, but maybe I got confused. In any case, this whole instance check in the `watchdog` library is problematic at best. This str conversion there will fail under Python 2 with a non ascii base folder, so this isn't an option either. My approach to the solution was to monkey patch the library to use a backported `scandir` library under Python 2 as well, effectively avoiding the whole `listdir` use and associated issues altogether.

I'll have to check if this indeed fails with *any* file in the `watched` folder, not just non-ascii stuff, in which case this might merit inclusion in the hotfix release I will likely have to do anyway for #3678

commit ad623007 (2020-08-05): 🐛 Work around for a watchdog Py2 bug

If the PollingObserver is used under Py2 on a directory that contains
utf8 file names, the default listdir implementation (scandir isn't
natively available under Py2) will generate unicode instead of str. Due
to an insufficent instance check in watchdog that will lead to a
misidentification of an entry as an DirEntry object instead of a str,
and the attempt to read the name property from that will cause an
AttributeError.

We monkey patch watchdog to use our bundled scandir backport where
needed instead of trying to fix the instance check, as that will per [... cut]
commit 89412a76 (2020-08-06): 🐛 Work around for a watchdog Py2 bug

If the PollingObserver is used under Py2 on a directory that contains
utf8 file names, the default listdir implementation (scandir isn't
natively available under Py2) will generate unicode instead of str. Due
to an insufficent instance check in watchdog that will lead to a
misidentification of an entry as an DirEntry object instead of a str,
and the attempt to read the name property from that will cause an
AttributeError.

We monkey patch watchdog to use our bundled scandir backport where
needed instead of trying to fix the instance check, as that will per [... cut]
closed 2020-08-06 by foosel