OctoPrint/OctoPrint #3678: Plugin Installation via URL does not seem to work on OctoPrint 1.4.1

osubuu, 2020-08-05:
#### What were you doing?

I was trying to install a plugin via URL using the plugin manager.

Exact steps to re-create the bug:
1. Go to Settings > Plugin Manager > Get More
2. Copy and paste the following URL into the **"... from URL"** input field: https://framagit.org/razer/octoprint_z_probe_offset/-/archive/latest/octoprint_z_probe_offset-latest.zip
3. Error is shown on screen (see screenshot below)

I have troubleshooted the issue and have a [proposed fix here](https://github.com/OctoPrint/OctoPrint/pull/3677).

#### What did you expect to happen?

For the plugin to be installed successfully.

#### What happened instead?

The error messaged (screenshot below) showed up instead.

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.4.1

#### Operating System running OctoPrint

MacOS

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

_octoprint.log installation error_
[image]

_plugin_pluginmanager_console.log installation error_
[image]

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2020-08-06:
Argh, thanks for reporting this. I must admit I only tested installing plugins from URL that are hosted on Github, and those send `filename` in `Content-Disposition` without enclosing quotes. 

Will have to decide whether this requires a hotfix release.

foosel, 2020-08-06:
I merged your PR into a hotfix branch, but then got thinking... Parsing this header with regex now seems like an exceptionally bad idea, and after some searching I found that `werkzeug.http.parse_options_header` is built exactly for stuff like this, so I've reworked the implementation accordingly. 

The URL from above and Github URLs now both work flawlessly on my end, but could you verify that this is the case for you too? https://github.com/OctoPrint/OctoPrint/tree/hotfix/3678

Currently strongly leaning towards a hotfix release, as this could backfire badly should Github suddenly decide to change their behaviour too. Monday the latest, possibly still today.

commit 14a5f89b (2020-08-06): 🐛 Fix download filename parsing

Closes #3678
closed 2020-08-06 by foosel