OctoPrint/OctoPrint #3680: [Request] Add further information to the Print State section

Sophist-UK, 2020-08-06:
**Is your feature request related to a problem? Please describe.**

Not really a problem as such - but more of a nice to have.

**Describe the solution you'd like**

It would be useful to have some addition information displayed in the Print State section of the Octoprint windows showing:

a. Layer x of y
b. The last M117 message sent to the printer's display.

**Describe alternatives you've considered**

Travelling to the remote printer and looking at the display rather than simply looking at the Octoprint screen.

fieldOfView, 2020-08-06:
This is what plugins are for, eg: https://plugins.octoprint.org/plugins/status_line/ in combination with https://plugins.octoprint.org/plugins/detailedprogress/

closed 2020-08-06 by fieldOfView

Sophist-UK, 2020-08-06:
Apologies - I had looked through the plugins but missed these.