OctoPrint/OctoPrint #3683: [1.14.2] Octoprint will hard crash if disconnect or connect is sent during an process

Kokorone, 2020-08-08:
#### What were you doing?

1. SSH to your pi, this will get the kernel crash error output in the terminal window at the end of the steps.
2. Send a command that takes a second to process, such as m997
3. Click disconnect after while it's still processing
4. Observe the pi freezing in the next 30 seconds, and a error sound. Reopen the ssh terminal, and you'll see the kernel crash output.

#### What did you expect to happen?

Octoprint continues to work as expected

#### What happened instead?

Octoprint hard crashes and requires a reboot of the entire pi to resolve

#### Did the same happen when running OctoPrint in safe mode?

Same crash happens in safe mode

#### Version of OctoPrint

1.14.2

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Artillery X1 with a SKR 1.4 Turbo. Marlin 2.0.6

#### Browser and version of browser, operating system running browser

Chrome, latest version, windows 10

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.410951] Internal error: Oops: 206 [#1] SMP ARM

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411335] Process kworker/0:2 (pid: 1366, stack limit = 0x8f51514e)

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411349] Stack: (0xef92bdd0 to 0xef92c000)

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411362] bdc0:                                     00000000 00000c00 ef92bdf4 ef92bde8

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411380] bde0: c0a557e8 c03e46ac ef92be3c d6f62b40 c1204f88 ef92be18 00000000 ef92be68

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411398] be00: 00001388 d65ddb80 ef92be54 ef92be18 c07ea8b0 c07e9bb0 00000000 00000000

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411415] be20: ef92be20 ef92be20 00000000 1bbb67a5 c1204f88 d649d800 00000000 00000000

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411433] be40: 00000000 00000001 ef92be9c ef92be58 c07eaa08 c07ea850 ef92be7c 00000000

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411450] be60: c9980000 00000002 00000000 1bbb67a5 ef92be9c 00000082 00000000 d649d800

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411479] be80: d6498800 d6498b48 d6498d04 00000000 ef92bed4 ef92bea0 c07eb830 c07ea940

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411497] bea0: 00000000 00000082 00000000 00000000 00001388 1bbb67a5 c023ea8c 00000010

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411514] bec0: d6498d14 d6498804 ef92befc ef92bed8 bf100ac8 c07eb7e0 d6498d14 d809a900

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411531] bee0: eff13340 eff16600 00000000 00000000 ef92bf34 ef92bf00 c023e224 bf1009c8

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411548] bf00: eff13340 eff13340 00000008 eff13340 d809a914 00000008 eff13358 c1203d00

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411565] bf20: ef92a038 d809a900 ef92bf7c ef92bf38 c023ea8c c023e0b4 ef92bf5c 00000000

Message from syslogd@OctoPi4 at Aug  8 09:08:21 ...
 kernel:[ 1084.411582] bf40: c1203d00 eff13358 c1203d00 c12a32a7 ef92bf7c d6416840 d6d3bec0 00000000

#### Link to contents of Javascript console in the browser

Browser no longer loads, so no java to mention

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-08-08:
To be honest, that sounds rather like a driver crash caused by the M997 on OS level, so nothing that OctoPrint could ever hope to handle. 

Also: logs please, they aren't optional.

Kokorone, 2020-08-08:
> To be honest, that sounds rather like a driver crash caused by the M997 on OS level, so nothing that OctoPrint could ever hope to handle.
> 
> Also: logs please, they aren't optional.

The boards themselves are working as expected after the m997. Zero issues printing or connecting to other usb interfaces. Once the pi has been hard rebooted, it is able to connect to the printer board without any issue.

 I can provide logs in the morning since I’ve tested the crash across three different printers with reliable success in reproducing. One is using lpc1768, the other is lpc1769, and the last is stm32f1. All of them are still 100% operational after octoprint completely crashes.

Kokorone, 2020-08-08:
Attached are the serial.log and octoprint.log.

There isn't much to see in octoprint.log since it doesnt see anything during or after the crash, it only begins logging after the reboot.

In serial.log, you can see the lack of normal disconnection output in terminal on the second attempt. The first attempt was successful without a system crash. 

Please refer to my first post for the ssh context on the kernel crash itself.

[octoprint (3).log](https://github.com/OctoPrint/OctoPrint/files/5045514/octoprint.3.log)
[serial (1).log](https://github.com/OctoPrint/OctoPrint/files/5045515/serial.1.log)

foosel, 2020-08-08:
What happens if you run octoprint manually from your SSH session and reproduce the issue?

I'd hope some error would be logged to console at least then:

    sudo service octoprint stop
    ~/oprint/bin/octoprint serve --debug

foosel, 2020-08-08:
Also, is there anything more maybe in `/bar/log/syslog` or the output of `dmesg`?

Kokorone, 2020-08-08:
> What happens if you run octoprint manually from your SSH session and reproduce the issue?
> 
> I'd hope some error would be logged to console at least then:
> 
> ```
> sudo service octoprint stop
> ~/oprint/bin/octoprint serve --debug
> ```

attached is the debug log
[debuglog.txt](https://github.com/OctoPrint/OctoPrint/files/5045633/debuglog.txt)

foosel, 2020-08-10:
This is going to take a while as I first have to get my hands on/update hardware with which to possibly be able to reproduce, and I have a much needed vacation coming up.

If anyone else can reproduce this in the meantime, I'd be really interested in `dmesg` and `syslog`, possibly via an additionally connected tty on the pi itself, and a `git bisect` between 1.4.0 and 1.4.2 to determine the commit that introduced the change in behaviour, if that can be pinpointed through reliable enough reproduction. Would also be interesting to know if that even *can* be reproduced by anyone else.

OctoPrint is still simply using pyserial to talk to the printer, and error handling on connection reset there hasn't changed. pyserial has not been updated in 1.4.1/1.4.2 either compared to 1.4.0 (3.4 has been the current version since 2017).

Kokorone, 2020-08-26:
> This is going to take a while as I first have to get my hands on/update hardware with which to possibly be able to reproduce, and I have a much needed vacation coming up.
>
> If anyone else can [...]

I've done some more deep diving into this issue. OctoPi itself will never run into this issue if you exclusively update your pi through octoprint. I usually regularly run full system updates on my pi via ssh, and it appears that something in the newer kernels pushed through causes the hard crash problem. This might be worth looking into for the future when the Octopi image is updated or there will be a problem with kernel crash again since it has not been fixed in recent kernel updates. 

For example, I'd run every few weeks

sudo apt-get update
sudo apt full-upgrade

foosel, 2020-09-07:
@guysoft this is probably something you should keep an eye out for on your end as well. Apparently there's an issue with current RPi kernels. Currently not part of the pulled in base image, but that might change at some point. Not sure how to proceed here from my end tbh, apart from recommending people not install updated kernels, which doesn't really seem a good idea. 

Is there an upstream ticket for that on the RPi tracker, @Kokorone? If not, may I suggest you [create one](https://github.com/raspberrypi/linux)?

foosel, 2020-10-08:
Closing as this is more an OS issue than an OctoPrint issue.

closed 2020-10-08 by foosel

Kokorone, 2021-02-06:
@foosel Looks like this was just mitigated in marlin bugfix 2.0.x. Here is the link to the commit.

[https://github.com/MarlinFirmware/Marlin/commit/004bed8a7fc3ff9feb73a0ea9794635b50073c27](url)

cp2004, 2021-02-10:
Upstream bug report in the Raspberry Pi kernel: https://github.com/raspberrypi/linux/issues/4120