OctoPrint/OctoPrint #3685: Elevated CPU temps and utilisation following 1.4.2 update on Pi 4 4GB

morphias2004, 2020-08-09:
#### What were you doing?

Upgraded to 1.4.2 from 1.4.0

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

Improved functionality, etc

#### What happened instead?

Raspberry Pi4 CPU temps now consistently ~15 degree celcius higher. CPU usage at idle is around double as well. Was typically 6% when idle, but now site at 12-15%. Temps used to site around 54C doing a print, now sits at 69-71C whilst doing nothing.

Running a Pi4 Model B Rev 1.2 4GB

#### Did the same happen when running OctoPrint in safe mode?

Yes. Does drop the temp about 4C when doing nothing.

#### Version of OctoPrint

1.4.2

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Pusa i3 MK3S

#### Browser and version of browser, operating system running browser

Windows 10 x64
Chrome 84.0.4147.105 x64

#### Link to octoprint.log

https://github.com/morphias2004/Octoprint-Logs

#### Link to contents of terminal tab or serial.log

https://github.com/morphias2004/Octoprint-Logs

#### Link to contents of Javascript console in the browser

https://github.com/morphias2004/Octoprint-Logs

#### Screenshot(s)/video(s) showing the problem:

I note when running a 'top' on the pi, kworker/u8:3-brcmf_wq/mmc1:0001:1 is generally at the top around 15% CPU.

I've uploaded a number of screen captures.

https://github.com/morphias2004/Octoprint-Logs

[image]
[image]

I have read the FAQ.

YES

Boerner1337, 2020-08-10:
I am running a Pi 3 B+ model and don't see this issue with 1.4.2 

[image]

My CPU load is somewhere between 4% and 7% while printing. Also did not notice any changes in temps. 

Greetings, 
Kev

foosel, 2020-08-10:
I cannot report any kind of abnormal CPU load either, nor increased `kworker` usage. 

[image]

This is from an idle system connected to a printer. Can't do test prints right now or my office will reach melting temperatures even with AC, but I didn't notice anything during RC phase either (and I was on the lookout since the first RC *did* have a CPU usage issues on the `octoprint` process).

`kworker/u8:3-brcmf_wq/mmc1:0001:1` is part of the kernel, not OctoPrint itself. I'm pretty sure that's actually the wlan driver and it's related to data transfers going over the wifi interface (that looks mightily familiar from my time debugging the [performance issues observed on the 0w on bandwidth usage of any kind](https://github.com/guysoft/OctoPi/issues/318#issuecomment-284771956)). Is this with the webcam active? Any changes when disabling octodash?

morphias2004, 2020-08-10:
Thanks for coming back to me so quickly. I am a new to the whole 'Pi' experience, so I appreciate everyone's patience and support. 😄 

The printer and Pi have been powered off completely for 36hrs. I just fired the Pi up after seeing the replies to this and the CPU is sitting around 47-50C. I've then installed the Resource Monitor plugin so I could see more of what was going on - awesome plugin BTW! 👍 

After about 10mins, the temp was still stable at 47-48C.

I then powered on the printer to get OctoDash to wake up and have not seen any increase in temp or CPU utilization.

What I did notice in Resource Monitor was constant WLAN traffic upload of ~43MB/s. I quickly realised that was the webcam. The camera is streaming 1920x1080 @ 15FPS, so the traffic is expected. I unplugged it to see what difference it made.

Here is the stats without the webcam - CPU drops to about 3-4%:

[image]
[image]

And with the webcam - CPU at 7-12% and temps stabilised at ~54C, which is where it was prior to the update:

[image]
[image]

My guess is that the system just needed a full power cycle following the update - the good old "Have you tried turning it off and on again?" 🤷  No amount of rebooting or restarting from the menus got it to clear the other day.

Anyway, I'm happy to have this one closed out and thank-you again for coming back to me so quickly.  👍

foosel, 2020-08-10:
Perfect, closing this then. Glad it cleared up, thanks for the pleasant interaction!

closed 2020-08-10 by foosel