OctoPrint/OctoPrint #3686: [1.4.2] Fields  "ExtrusionLength", "Feed rate Modifier" and "Flow rate Modifier" on the "Control" tab are displayed empty

EagleB3, 2020-08-10:
Updated version 1.4.0 to 1.4.2. After starting Octoprint On the Control tab, the "ExtrusionLength", "Feed rate Modifier" and "Flow rate Modifier" fields are displayed empty. The fields remain empty with all plugins I use, and in safe mode (the screenshot is attached).
The problem has been confirmed by multiple users (https://www.facebook.com/groups/piprint/permalink/984235615348495/)
I am using Windows 10 and Mozilla Firefox v.79.0 x64.

I can select and provide additional logs and screenshots, if necessary.
[image]

P.S. As far as I can see, the ability to set a custom value for the default extrusion value in the printer profile was implemented in 2014 ([issue #513](https://github.com/OctoPrint/OctoPrint/issues/513)). But this is not possible now. It's a pity...

cp2004, 2020-08-10:
**This is not a bug, or a problem.** Those values have been removed from the UI for the following reasons:
1. There is no universal support for querying that number from the printer, so it caused more headaches as often the value was wrong. Those values can be adjusted from the printer screen, but OctoPrint has no way of knowing what it changed to. This lead to lots of users complaining 'The number was wrong!'
2. In firmware that does support querying those, they still do not report back if it is changed on the printer screen. So regularly query, you may say? However there are two issues with this, (1) Constantly querying saturates the serial connection, resulting in lower performance, especially when many people struggle with getting a serial connection set up. (2) There is no way of knowing whether the firmware supports querying, and just blindly firing off commands is a VERY bad idea, it could have disasterous consequences. Shall we say, a new firmware comes along, but the same command means 'Heat to 2000C'? Obviously exaggerated, but you should get the point.

From the release notes 
>#3569 - Properly reflect on the UI that feed rate and flow rate modifiers are write-only ephemeral values that cannot be read back from the firmware by removing permanent display.

cp2004, 2020-08-10:
However I am inclined to agree, that the extrusion length being missing is quite unclear.

EagleB3, 2020-08-11:
With an empty field, I press the button ("Retract", "Extrude", "Set flow rate" or "Set feed rate") and the corresponding command with a specific value is sent to the printer. So, **THIS** value **MUST** be displayed in the field.
By the way, the values ​​in the commands are sent quite correct, they correspond to the traditional default values ​​- 5mm, 100% and 100%. So this is not a problem either.
How this value was obtained to fill in the field (from the printer's response, from the default settings, or from the user's printer profile settings) is a separate question, and this is not a problem.

But the number given in the command does not correspond to the current value in the field - **this is a bug**.

foosel, 2020-08-11:
The missing extrusion length is a bug, seeing it here as well. I changed the extrusion amount to be a number instead of a text field, and apparently placeholders (which is what the "prefilled" value was) aren't supported on that. I'll fix that.

Flow and feed rate "missing" were already explained, and that's also what @cp2004 was talking about as "not a bug, or a problem" @EagleB3. That change was also documented in the [1.4.1 release notes, linked to from the 1.4.2 hotfix release notes](https://github.com/OctoPrint/OctoPrint/releases/tag/1.4.1) (incl a link to the corresponding ticket), which apparently next no one seems to read. I get it, they can be long, but **it would be lovely if they at least were read once an unexpected change in behaviour is observed** as it would save everyone a ton of time of either (apparently even collectively) scratching their heads or constantly having to quote them. I write them (which btw takes hours) so all intentional changes are documented and people can check what actually goes into a release. So please do me the favour to at least read them before opening bug reports for intentional and explained changes.

cp2004, 2020-08-11:
@foosel Might also be worth disabling the button when there is nothing in the flow/feedrate boxes, as the button still sends 100% when there's nothing there

EagleB3, 2020-08-11:
> @foosel Might also be worth disabling the button when there is nothing in the flow/feedrate boxes, as the button still sends 100% when there's nothing there

@foosel, Yes, in terms of fields "Flow rate" and "Feed rate" I mean exactly that.
If the field is currently **null**, then it is wrong to send "M220 S**100**"/"M221 S**100**".

@cp2004 's proposed solution is very good! Making the "Set" button unavailable while there is no user-entered value in the field "Flow rate"/"Feed rate" is a **very good option**. This solution fits perfectly into the overall concept of the interface:
[image]

foosel, 2020-08-11:
> Yes, in terms of fields "Flow rate" and "Feed rate" I mean exactly that.

Then you should have said so ;)

In any case, I just pushed 5ca20f741a89c070395b4adac417d55f2c92deec to solve that too. THe extrusion length issue has already been solved via 4d82f2a4c6b93b238343644f7d49e384e13a7f45. Available on the `maintenance` branch and ready for 1.5.0.

EagleB3, 2020-08-11:
@foosel, Thank you very much!

P.S. I'm sorry, but I don't know English well enough. And google's automatic translation is far from ideal... :(

foosel, 2020-08-11:
All good, just caused a bit of confusion here :)

commit 4d82f2a4 (2020-08-11): 🐛 Fix default extrusion length display

Closes #3686
closed 2020-11-30 by foosel