OctoPrint/OctoPrint #3691: 1.4.2 _ Uploading and deleting gcodes won't terminate

cagilundar, 2020-08-15:
Hi,
After latest updates, I am having gcode file uploading and deleting issues. If I want to upload, status changes to  _saving_ and  freezes.  I reconnect to octoprint and I see file there. Same happens with deleting with relative cases. 
[image]

fieldOfView, 2020-08-15:
I don't know if you meant to close the issue, but disabling the gcodeeditor plugin should fix this problem.

cagilundar, 2020-08-15:
> I don't know if you meant to close the issue, but disabling the gcodeeditor plugin should fix this problem.

Thanks a lot. It was marked as imcomplete ticket. This is my first request (post) on git hub, I've probably missed some mechanics.

fieldOfView, 2020-08-16:
If you have the GcodeEditor plugin installed, please disable it.

For future reference, when the GitIssueBot complains about missing information in your issue, that is an invitation to add that missing information, not to close your report. It is important that issues get reported, or else bugs don't get fixed. 

In this case the issue has already been reported in https://github.com/OctoPrint/OctoPrint/issues/3650, and the problem has been identified as a bug in a specific plugin (GcodeEditor). So I'll close your issue again and mark it as a duplicate. If my diagnosis is not correct (eg if you don't have the GcodeEditor plugin installed), please let me know and we'll investigate further.

closed 2020-08-16 by fieldOfView

cagilundar, 2020-08-16:
Hey _fieldofview_. It's the same problem with [#3650](https://github.com/OctoPrint/OctoPrint/issues/3650).Sorry for duplicate. I was thinking it was related with 1.4.2 but apparently it existed in previous releases (meaning 1.4). Thanks for the tip. I'll disable GCodeEditor and give it a try.